nightscout/AndroidAPS #2792: [DEV] AAPS/AAPSClient NSV3-tab getting alot of spam (V3+WS API)

olorinmaia, 2023-09-17:
AAPS/AAPSClient 3.2.0-dev-l
Build: f717d7a0b2-2023.09.16

Not sure what triggered it, but last operation I made was enacting TT from AAPSClient around 15:55. Then checked logs and there is alot of spam in both AAPS/AAPSClient NSV3-tab. As a consequence sync isnt working.

Screens: 
AAPSClient:
[image]
[image]

AAPS:
[image]
[image]

Logs AAPS:
[AndroidAPS_LOG_1694959105301.log.zip](https://github.com/nightscout/AndroidAPS/files/12642979/AndroidAPS_LOG_1694959105301.log.zip)
Logs AAPSClient:
[AndroidAPS_LOG_1694959080007.log.zip](https://github.com/nightscout/AndroidAPS/files/12642981/AndroidAPS_LOG_1694959080007.log.zip)

Alamo04, 2023-09-18:
Confirmed.

Old issue comes up again = Profile Spamming

[AndroidAPS_LOG_1695038502416.log.zip](https://github.com/nightscout/AndroidAPS/files/12647544/AndroidAPS_LOG_1695038502416.log.zip)

NighthawkEvP, 2023-09-22:
also confirmed

MilosKozak, 2023-09-23:
https://github.com/nightscout/cgm-remote-monitor/issues/8096

MilosKozak, 2023-09-25:
https://github.com/nightscout/cgm-remote-monitor/pull/8100

parapenT1sta, 2023-09-25:
> [nightscout/cgm-remote-monitor#8100](https://github.com/nightscout/cgm-remote-monitor/pull/8100) this resolve the issue

When will be merged? Do you have an idea?
CGM Monitor is slower than AAPS repo :-)

MilosKozak, 2023-09-25:
@bewest is maintaining NS repo now

closed 2023-10-03 by MilosKozak