nightscout/AndroidAPS #2796: [DEV] AAPS IOB differ from AAPSClient (tested with v1 API)

olorinmaia, 2023-09-18:
AAPSClient 3.2.0-dev-l
Build: f717d7a0b2-2023.09.16
Remote: https://github.com/nightscout/AndroidAPS
Flavor: aapsclientrelease
Nightscout versjon: 15.0.0

See attached screenshots. Seems like a temp basal at 17:04 is the culprit.

**Logs:**
[AndroidAPS_LOG_1695056440606.log.zip](https://github.com/nightscout/AndroidAPS/files/12651529/AndroidAPS_LOG_1695056440606.log.zip)

**Screenshot AAPSClient:**
[image]
[image]

**Screenshot AAPS:**
[image]
[image]

olorinmaia, 2023-09-20:
Happend again.
Logs AAPS:
[AndroidAPS_LOG_1695239877826.log.zip](https://github.com/nightscout/AndroidAPS/files/12677159/AndroidAPS_LOG_1695239877826.log.zip)
Logs AAPSClient:
[AndroidAPS_LOG_1695236772114.log.zip](https://github.com/nightscout/AndroidAPS/files/12677161/AndroidAPS_LOG_1695236772114.log.zip)

Time: 20:30
AAPSClient:
[image]
AAPS:
[image]

olorinmaia, 2023-09-20:
I know I changed from v3 APi to v1 API before I noticed it, if that's whats causes it then it will probably not be normal behaviour as most people will stick with one type of api until forced onto the other.

MilosKozak, 2023-09-23:
the record has wrong duration. what is in NS?

MilosKozak, 2023-09-23:
https://github.com/nightscout/AndroidAPS/commit/8c6157b8f6219bd826543393a9cc570bab7b23bb

closed 2023-09-23 by MilosKozak