nightscout/AndroidAPS #2815: [DEV] Autosens is shown on Overview even if disabled

justmara, 2023-09-21:
[This](https://github.com/nightscout/AndroidAPS/blob/dev/plugins/main/src/main/java/info/nightscout/plugins/general/overview/OverviewFragment.kt#L1091) line of code makes aaps always draw Autosens data on overview screen as 'active', even if isAutosensModeEnabled() returns false.

[image]
[image]

NighthawkEvP, 2023-09-22:
And if you select dynISF the related figures are missing since a few days. 
Could that be related to the Autosens thing?

MilosKozak, 2023-09-22:
this is by design because it's calculated anyway. it's there to notify user some change can be usefull

justmara, 2023-09-22:
@MilosKozak  There is special icon with `x`-mark that were used to show `disabled` autosens. Why it is not used anymore? The design has changed?

MilosKozak, 2023-09-23:
right! resolved in https://github.com/nightscout/AndroidAPS/commit/0ef3d9b3757cc3b983132cae832d04cb51180395

closed 2023-09-23 by MilosKozak