nightscout/AndroidAPS #2818: Nano disconnection

Korostensky, 2023-09-22:
The pump TouchCare Nano A8 is disconnected
--------------
Build: dev-l (personal branch) 
Pump: Medtrum TouchCare Nano A8
Time: ca 12:40 - 16:00 and 23:10 - 02:30
Logs: 
[AndroidAPS._2023-09-21_02-00-10_.2.zip](https://github.com/nightscout/AndroidAPS/files/12699940/AndroidAPS._2023-09-21_02-00-10_.2.zip)
[AndroidAPS._2023-09-21_22-57-29_.3.zip](https://github.com/nightscout/AndroidAPS/files/12699935/AndroidAPS._2023-09-21_22-57-29_.3.zip)
[AndroidAPS._2023-09-21_22-57-29_.4.zip](https://github.com/nightscout/AndroidAPS/files/12699936/AndroidAPS._2023-09-21_22-57-29_.4.zip)
[AndroidAPS._2023-09-22_02-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/12699939/AndroidAPS._2023-09-22_02-00-00_.0.zip)

The pump delivers only a standard basal dose of insulin and does not respond to changes in glycaemic values. I don't know the reason. Connection time expires but the device is still in the same place, no loss occurs due to long distance.

In most cases the pump is controlled by AAPS as standard, including the use of SMB.

[image]

MilosKozak, 2023-09-22:
this looks like PlayProtect issue

Korostensky, 2023-09-22:
Yes, play protect was on, you're right. Sorry for the unnecessary question, I could have thought of that too. I don't understand how it got set to active. I turned it off.

closed 2023-09-22 by MilosKozak