nightscout/AndroidAPS #2821: Unable to activate profile after switching to dev branch

orbarya, 2023-09-23:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem

### Time of issue -  2023-09-22 20:01
### Issue - 
After changing from master branch to dev AAPS 3.2.0-dev-l, 
I couldn't activate the profile. I tried clicking the ACTIVATE PROFILE button but still in the OVERVIEW tab I still saw NO PROFILE SET.

eventually I was able to fix the issue by duplicating the profile and then activating the new profile with name "LocalProfile1 Copy". Unfortunately I do not have the logs showing this.

Build: ccabee0399-2023.09.22

[AndroidAPS_LOG_1695402777338.log.zip](https://github.com/nightscout/AndroidAPS/files/12707212/AndroidAPS_LOG_1695402777338.log.zip)

MilosKozak, 2023-09-23:
did you see confirmation dialog?

orbarya, 2023-09-24:
> did you see confirmation dialog?

> did you see confirmation dialog?

orbarya, 2023-09-24:
Yes I believe I did see but not completely sure

tepidjuice, 2023-10-24:
I have the same bug with a confirmation dialog. I was able to activate a profile by copying one of the standard profiles. However I was not able to change from this profile.

All tested profiles worked in 3.2.0-rc1.

closed 2024-12-04 by MilosKozak