nightscout/AndroidAPS #2832: [DEV] BG in AAPSClient differ from AAPS/NS by decimal (v1 API)

olorinmaia, 2023-09-24:
AAPSClient 3.2.0-dev-l
Build: bdfa4a98fc-2023.09.24
Flavor: aapsclientrelease
Nightscout versjon: 15.0.0

Only applies to V1 API. in V3 BG is same in AAPS/AAPSClient and NS.

Logs AAPSClient:
[AndroidAPS_LOG_1695560769859.log.zip](https://github.com/nightscout/AndroidAPS/files/12708560/AndroidAPS_LOG_1695560769859.log.zip)

Screenshots: top images are AAPS, bottom AAPSClient.
[image]
NS:
[image]

MilosKozak, 2023-09-25:
I need log from boh sides

olorinmaia, 2023-09-25:
Is this what you need? from today:
AAPSClient:
[Uploading AndroidAPS_LOG_1695677050900.log.zip…]()

serverlog from ns.10be.de:
[ns.10be.de_serverlog.txt](https://github.com/nightscout/AndroidAPS/files/12719526/ns.10be.de_serverlog.txt)

Screens:
AAPS BG: (Same as NS)
[image]

AAPSClient BG:
[image]

olorinmaia, 2023-10-03:
I don't feel it's right to use time on this, since v1

closed 2023-10-03 by olorinmaia