nightscout/AndroidAPS #2837: [DEV] DynISF plugin seems to be not working

NighthawkEvP, 2023-09-25:
Since a couple of days there are no figures showing the result of the dynISF script compared to the profile ISF in the main screen (i.e. 67,9 -> 56,1).
Looking into the DynISF Tab the results of the plugin look different (Skript debug doesn't start with the dynISF version 2.0) and there ist no result other than the profile ISF.
I've got the impression that the DynISF plugin doesn't work at the moment? Is it running autosense plugin instead?

See two different log files from today with actual and former dev versions.

dev-l from 24.09.2023
[AndroidAPS_LOG_1695635047011.log.zip](https://github.com/nightscout/AndroidAPS/files/12713843/AndroidAPS_LOG_1695635047011.log.zip)

dev-l from  15.09.2023
[AndroidAPS_LOG_1695635291286.log.zip](https://github.com/nightscout/AndroidAPS/files/12713815/AndroidAPS_LOG_1695635291286.log.zip)

MilosKozak, 2023-09-25:
then you don't have enough TDD data. see 1st paragraph on DynISF tab

jamorenoPOP, 2023-09-25:
I have updated my dev-I branch because I bought a Smart watch, I have created the two apps, the fact is that since yesterday I installed the app I do not have dynamic ISF, it is active but it does not change the sensitivity as it did before. I have a week's worth of TIR data, nothing was lost in the app update. I have met all the objectives even on the 11th.
The build was on 09/23/2023 3.2.0-dev-I ccabee0399.
I have completed objective 11 and until Friday everything was working ok

NighthawkEvP, 2023-09-25:
> then you don't have enough TDD data. see 1st paragraph on DynISF tab

No, that is not the reason as I used it for several and didn't reset the db. I think it is looking for the wrong plugin.
See screenshots:
[image]
[image]

jamorenoPOP, 2023-09-25:
I cannot attach documentation now since my son is at the Institute, but I have a 7-day TIR data and since the update it has been 3 days of feeding and it is not activated.

hemmann, 2023-09-25:
The same here. Objective 11 is shown as not finished yet (in the DynISF log and in the Objective context menu), although DynISF, SMB etc. is activated in the settings.  Also, the DynISF adjustments are not visible in the status bar. Possibly there is a relation to the use of Libre 3, because you cannot use SMB by default.

closed 2023-09-25 by MilosKozak