nightscout/AndroidAPS #2839: [DEV] Crash on trying to enter minutes in TT window

jbr7rr, 2023-09-26:
Build: 92558e9d65-2023.09.25
Time: 08:21
Log: [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12723522/AndroidAPS.log)

App crash when trying to enter duration in temp target dialog

How to reproduce:

- Open TT dialog
- Press the minutes field, so keyboard will pop op
- App will crash

NighthawkEvP, 2023-09-26:
same here, app crashes by entering the first digit

hemmann, 2023-09-26:
here are the screenshots

[image]
[image]
[image]

parapenT1sta, 2023-09-26:
> mine don't crash android version?

Mine is also OK.

MilosKozak, 2023-09-26:
tested on 
Androd12 (digital clock appear - keyboard popup)
Android 13 (analog clock appear - no keyboard)
no crash :(

jbr7rr, 2023-09-26:
Pixel 6Pro android 13 > Crash
Redmi Note with custom rom android 11 also crash.

Will see if I can create stacktrace later today

NighthawkEvP, 2023-09-26:
in my case it is on a Pixel 5 with Android 13 and it seems to be in every case where a numeric keyboard is used. I tried TT and bolus assistent and both did crash. I went back to the version from 15.09.2023.

jbr7rr, 2023-09-26:
Stacktrace:
```

MilosKozak, 2023-09-26:
try to update
Androd java lib was not properly marked @Nullable

jbr7rr, 2023-09-26:
Jup, works for me! Thx!

closed 2023-09-26 by MilosKozak

NighthawkEvP, 2023-09-26:
works like a charm!
Thx Milos!