nightscout/AndroidAPS #2843: Passing Objective 11 requires using DynamicISF for at least 30 days (suggestion: ignore #11 unless plugin is active)

vanelsberg, 2023-09-26:
**AndroidAPS 3.2.0-dev-l**
**Flavor**: fullrelease

**Observation:**
Object 11 is shown as unfinished, can't complete/verify because not using (or planning to) DynamicISF

**Suggestion**:
Only enable/show object 11 when the DynamicISF plugin is activated as in ignore/hide when not using DynamicISF

MilosKozak, 2023-09-26:
You need only start it. And it's recommended to do it.
If more objectives come you'll need to have all previous finished

vanelsberg, 2023-09-26:
Think I did. But it stopped counting at 14 days when I stopped using DynamicISF?
Will recheck to see what happens when I temporarily enable DynamicISF

[image]

vanelsberg, 2023-09-26:
> If more objectives come you'll need to have all previous finished

Yes. That why I created this issue....
Will let you know if I can get it counting toward the 28 days...

swissalpine, 2023-09-29:
Counted to 28 without enabling dynISF. I see no issue.

robertrub, 2023-10-02:
I turned ON DynISF then started Obj11. When back to OpenSMB. 

Thought checking where it was and it had counted around 14 days.

Totally forgot about it and "verify"ed it around 45 days (with OpenSMB active and DynISF still not in use) and I had it "completed".

OTOH, didn't quite understand what was the Obj11's limitation... I could use DYNISF without Obj11 being validated. What changes if Obj11 is validated or not ?

MilosKozak, 2023-10-31:
@robertrub it's for new users. they have to stay 1 month on SMB before they can start DynISF

closed 2023-10-31 by MilosKozak