nightscout/AndroidAPS #2846: [DEV] AAPSClient occasionally misses to download SGV

szantos, 2023-09-28:
AAPSClient 3.2.0-dev-l d379b769d3-2023.09.14 with NSClientV3 + websockets
NS dev v15: d602de1
[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/12746351/Log.Export.txt)

AAPSClient misses to download or backfill some SGV - which is present in NS.

[image]

olorinmaia, 2023-09-28:
I recommend updating to latest dev and use v1 api until this issue is sorted
https://github.com/nightscout/AndroidAPS/issues/2792

szantos, 2023-09-29:
Thank you! Does switching between v1 and v3 make any issues?

olorinmaia, 2023-09-29:
Its no problem to switch between! But update dev first since there was a few bugs regarding that + more that has been fixed recently. When NS15 v3 api has been updated with fix Im going to test more as i believe there still are issues with new api and sync between AAPS, AAPSClient and NS. 😀

olorinmaia, 2023-10-04:
@szantos v3 API should be OK now, update to latest NS dev and see if you can reproduce this with v3

szantos, 2023-10-05:
Thank you! I'll test and tell if there are any issues.

MilosKozak, 2023-10-09:
create new issue if needed

closed 2023-10-09 by MilosKozak

olorinmaia, 2023-10-09:
Make sure NS is updated to latest dev, if that is true and this still happens create a new issue with logs!

ohjay73, 2023-10-09:
> Make sure NS is updated to latest dev, if that is true and this still happens create a new issue with logs!

I updated using todays commits and still same issue. New issue created #2890