nightscout/AndroidAPS #2848: [DEV-M] Widget is broken

swissalpine, 2023-09-28:
The widget is frozen after the update, re-adding (after deleting) is not possible.

robertrub, 2023-09-28:
I confirm. Widget is available but will not install. Android 13 S21FE
[image]

MilosKozak, 2023-09-28:
it works in emulator. what android version?

robertrub, 2023-09-29:
[image]

robertrub, 2023-09-29:
The video, if it helps...

https://github.com/nightscout/AndroidAPS/assets/76626474/51e511bc-1c33-401d-af9d-7eb9c444f925

swissalpine, 2023-09-29:
Samsung Z Flip 4, Android 13, One UI 5.1.1 (September 2023)
Same here: Fine with dev-l, buggy with dev-m
Logcat after installing (widget freezes):
```
08:39:24.024  I  addProviderLocked Provider is added Provider{ProviderId{user:0, app:10423, cmp:ComponentInfo{info.nightscout.androidaps/app.aaps.ui.widget.Widget}}} at 107
08:39:51.765  D  Provider info from ComponentInfo{info.nightscout.androidaps/app.aaps.ui.widget.Widget} won't be persisted.
```
Logcat after re-adding (only one line):
```
08:45:23.693  D  bindAppWidgetIdIfAllowed() appWidgetIds = 121, provider = ComponentInfo{info.nightscout.androidaps/app.aaps.ui.widget.Widget}
```
More info needed?
Thanks!

CaroGo, 2023-09-29:
same here with Samsung Fold 4 , Android 13

MilosKozak, 2023-09-29:
hmmm it does work in emulator :(

robertrub, 2023-09-30:
According to a book I read recently, we might all be the products of a huge emulator... ;)

The widget is working with this version.

[image]

Something changed between that and last Dev.

Josch24, 2023-09-30:
I installed the last Dev-M. On my Pixel 4a the Widget is working. I am also on Android 13.

robertrub, 2023-09-30:
Maybe it is Samsung's ONE UI 5.x related... Both of us with problems are on Samsung phones.

MilosKozak, 2023-09-30:
you may try to uninstall aaps

robertrub, 2023-09-30:
Uninstalling will loose too much data.

Tested again with dev-l and the widget installed

[image]

HuismanErik, 2023-09-30:
I did a reinstall with this change https://github.com/nightscout/AndroidAPS/pull/2850 and it fixed the Widget

ninelore, 2023-09-30:
For the record: This doesnt happen on my Pixel 6 Pro on commit 81548b7

swissalpine, 2023-09-30:
Fix for the issue: https://github.com/nightscout/AndroidAPS/pull/2850
Thanks a lot!
@HuismanErik

robertrub, 2023-09-30:
I confirm too. Including Eric's PR 2850 on Dev solved the widget problem. When updating from dev-L to dev-M + PR2850, the widget disappears but can be put back in place.

closed 2023-09-30 by MilosKozak

vanelsberg, 2023-10-02:
FYI: Running dev-m (commit 81548b7d).
On Samsung S22 Ultra (Android 13, One UI 5.1): No problems with Widget.