nightscout/AndroidAPS #2853: BUG / DEV-M / DASH History restarts AAPS

robertrub, 2023-10-02:
When I look at the DASH history, AAPS restarts (I see the AAPS icon and then the main page with the progress bar etc). First found by @ohjay

[image]

[image]

TheNeskik, 2023-10-02:
Same issue here
[AndroidAPS_LOG_1696248292700.log.zip](https://github.com/nightscout/AndroidAPS/files/12782465/AndroidAPS_LOG_1696248292700.log.zip)

TheNeskik, 2023-10-02:
Same here : log below
[AndroidAPS_LOG_1696250330973.log.zip](https://github.com/nightscout/AndroidAPS/files/12782791/AndroidAPS_LOG_1696250330973.log.zip)

robertrub, 2023-10-03:
To all that have commented and added logs.

Please try to use pod history again. It started working for me but I did nothing new... I didn't start a new pod. I just rebooted the phone but no idea if this is related. Now, pod history works.

The bug got solved all by itself... Not normal.

parapenT1sta, 2023-10-03:
> To all that have commented and added logs.
> 
> Please try to use pod history again. It started working for me but I did nothing new... I didn't start a new pod. I just rebooted the phone but no idea if this is related. Now, pod history works.
> 
> The bug got solved all by itself... Not normal.

A fix is waiting to be merged: https://github.com/nightscout/AndroidAPS/pull/2858

TheNeskik, 2023-10-03:
> To all that have commented and added logs.
> 
> Please try to use pod history again. It started working for me but I did nothing new... I didn't start a new pod. I just rebooted the phone but no idea if this is related. Now, pod history works.
> 
> The bug got solved all by itself... Not normal.

Not on my phone :)

Tried the fix and everything seems to be back in place.

ohjay73, 2023-10-03:
powercycled and still have the same crash experience.

jbr7rr, 2023-10-04:
> A fix is waiting to be merged: #2858
Fix needs testing. Have not tested it on a active pod. So if any of you can test the PR and see if everything is there in the pod history that would be really helpful

robertrub, 2023-10-04:
Ok, tested the PR of @jbr7rr and everything is working ok for me. Pod history is accessible (but that had started working for me before the fix) and recent events are recorded to history.

For me, the OR is a _go for merge_

closed 2023-10-04 by MilosKozak