nightscout/AndroidAPS #2854: NSClient History time-axis wrong scaling

ben80, 2023-10-02:
Problem
--------------
The time-axis in the history window is not scaled properly for all hourly-steps.
The time difference between the vertical grid lines is not always constant.

Examples - wrong grid
--------------
See screenshots attached.

Screenshot_20230927-062114.png: between 4 and 5 o'clock only one hour difference instead of 2
[image]

Screenshot_20231002-082624.png: between 7 and 10 o'clock 3 hours difference instead of 2
Screenshot_20231002-082746.png: between 8 and 12 o'clock 4 hours difference instead of 3

Examples - plot scaling seems to be ok
--------------

MilosKozak, 2023-10-04:
wasn't it day of DST change?

ben80, 2023-10-04:
No, that's what i first thought, also.
But DST change will be oct 29th 2023 only (in one month).

I think the plot is scaled correctly. 
But the ticks of the vertical grid lines are not equidistant within the graph.
Should be e.g.:  0 - 2 - 4 - 6 ... o'clock
But is:                0 - 2 - 5 - 7 ... o'clock

MilosKozak, 2023-10-09:
https://github.com/nightscout/AndroidAPS/issues/2729

MilosKozak, 2024-02-11:
it's a bug in library