nightscout/AndroidAPS #2860: Upload to tidepool crashes new upgrade AAPS 3.2-rc1

PedanticAvenger, 2023-10-03:
I've been waiting for tidepool support for a while. ;)

Just upgraded to rc1 and got everything else setup.  Plugin itemizes what needs to be uploaded, assembles it then right after upload begins the app crashes.  This has happened 4 times consecutively before I disabled tidepool.

I have a log and can attach it if needed/useful.
Build: dd6081efd2-2023.10.03
Flavor: fullrelease
Nightscout version: 15.0.0

[AndroidAPS._2023-10-03_15-54-04_.34.zip](https://github.com/nightscout/AndroidAPS/files/12797268/AndroidAPS._2023-10-03_15-54-04_.34.zip)

MilosKozak, 2023-10-04:
log doesn't contain any crash
replicate the issue, update with new logs and time of occurence
make sure the time is in logs

PedanticAvenger, 2023-10-04:
Will do.
I have previous logs as well, is a crash obvious in them if I go looking to verify?

PedanticAvenger, 2023-10-04:
Active and previous included as well as screen shot of the tidepool tab taken about a second and a half before crashing.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12807525/AndroidAPS.log)
[AndroidAPS._2023-10-03_22-20-35_.43.zip](https://github.com/nightscout/AndroidAPS/files/12807526/AndroidAPS._2023-10-03_22-20-35_.43.zip)
[image]

MilosKozak, 2023-10-09:
i cannot replicate the issue

PedanticAvenger, 2023-11-09:
AFter full release it appears to be behaving correctly so will close this.

closed 2023-11-09 by PedanticAvenger