nightscout/AndroidAPS #2861: Feature Request: Different DynISF Adjustment Factor for Different Hours of the Day

yhonatanb20, 2023-10-04:
I find in the recent time that I need an adjustment factor >100% to effectivly balance the day, but at night it causes some hypos. Tryed to get over it with targets and automations without sucsess.
I think it would be great if the adjustment factor would be set similarly to profile parameters, so that different adjustment factors could be set for different hours of the day (somewhat like ISF in SMB) (e.g. 113% from 9:00 and 100% from 23:00). I believe it makes much sense given that it's the main parameter affecting DynISF's result.

szantos, 2023-10-05:
DynISF with good parameters during the day used to be too much at night - something I've noticed, too.

Putting the adjustment factor into the profile would also fill the existing gap of ISF-scaling at a precentual profile change. Currently DynISF is not affected by a profile change - I hope this will be solved before a stable release.

Further please consider adding an automation action to set the DynISF adjustment factor.

justmara, 2023-10-08:
[this](https://github.com/nightscout/AndroidAPS/pull/2767) DynISF realization takes profile % into account (when 'Obey profile' knob is switched On).
I see two possible ways for this feature to be implemented:
1. (easy) Make DynISF follow profile % changes
2. (harder) Make DynISF adjustment value to be changeable via automation (afair currently there are no settings that can be changed via automation)

Some people find circadian being enough to deal with daily sensitivity change

Koeppchen, 2023-10-17:
> Another approach as a sidenote: After using DynISF for a few months, I've started to change the hardcoded TDD-mix to let it react more quickly. From the original weights and time-windows of 23% 0-4h + 10% 4-8h + 33% 1d + 34% 7d I've shifted to 33% 0-3h + 27% 3-6h + 25% 1d + 15% 7d over a period of more than a year in more little steps.
> 
Very Interesting. I will try to increase the influence of the last 8 hours a little bit. Is there a place where dynisf Ideas or changes are discussed?