nightscout/AndroidAPS #2873: Cannot start Objective 11 - buttons are missing

ledrexter, 2023-10-05:
I'm using dev compiled 6 days ago. My phone is Xiaomi 11T Pro. 
The buttons to start and verify the Objective are missing. Should I add log files as well?

[image]

[image]

[image]

jbr7rr, 2023-10-05:
Check if all.other objectives are passed. There are probably some open. Objective 3 got some new questions

ledrexter, 2023-10-05:
Thanks a lot!
All in Objective 3 were finished but there was a button Verify that I have missed when checking.
Closing this one!

closed 2023-10-05 by ledrexter