nightscout/AndroidAPS #2881: Medtrum pump driver default setting and concerns

tim2000s, 2023-10-07:
I have been testing the Medtrum Nano pump with 3.2 and have a couple of observations with its use.

1. The driver should default to the pump standard settings, ie, terminate after three days and 8 hours grace period. It has a 90° steel cannula, so exceeding this limit should be a decision that is made during set up not as a default, due to the trauma that can occur.

2. The driver doesn't yet seem to support the pump low reservoir and empty alerts. This can cause issues both overnight (which I experienced) and when someone goes out for the day, and probably should be addressed before it goes live in a master version.

jbr7rr, 2023-10-07:
Thanks for your feedback

1. In AAPS tubed pumps also don't have an alarm on canula age. and this has been discussed already in https://github.com/nightscout/AndroidAPS/issues/2404 I can ofcourse change the default making it more inline with default PDM/Medtrum app behavoir and will do that

2. A bit same as above. Also behavoir of the medtrum app is a bit spotty at this. If you leave the default behavoir (pump alarm on beep) the patch will beep below 20Units this is default behavoir from the patch and cannot be changed, Also it will show up in the alarm list on the Medtrum tab and this warning cannot be reset. I can add an additional warning on this alarm/popup in AAPS. But then do you want sound with this and be able to turn the sound off? This will complicate some things though..

jbr7rr, 2023-10-08:
https://github.com/nightscout/AndroidAPS/pull/2885

- Patch expiration enabled by default
- Option to show non critical pump alerts (enabled by default) including low reservoir warning at 20units.

Still in draft for testing, feel free to test

To reiterate: In current dev and RC1 you will get an alert from AAPS if pump is stopped due to Empty reservoir. if this is not the case please add logs.

Also note, that there are patch faults in which the pump completely stops working and will not communicate the error with connected device, there is nothing in driver we can do about that (other then the no communication alarm). This can happen for example when water leaks between the pump contacts and shorting the battery

tim2000s, 2023-10-08:
> #2885
> 
> * Patch expiration enabled by default
> * Option to show non critical pump alerts (enabled by default) including low reservoir warning at 20units.
> 
> Still in draft for testing, feel free to test
> 
> To reiterate: In current dev and RC1 you will get an alert from AAPS if pump is stopped due to empty reservoir (or any other error reason). If this is not the case please add logs.
> 
> Also note, there are patch faults in which the pump completely stops working and will not communicate the error with connected device, there is nothing in driver we can do about that (other then the no communication alarm). This can happen for example when water leaks between the pump contacts and shorting the battery

Last point I haven't seen so far, but I assume it's the same for the app. Will test the rest.

tim2000s, 2023-10-09:
> #2885
>
> * Patch expiration enabled by default
> * Option to show non critical pump alerts (enabled by default) including low reservoir warning at 20units.
>
> Still in draft for testing, feel free to [...]

Spoke too soon. Overnight last night I suffered the pump stopping working completely, for no apparent reason. 

I wonder whether it might be worth ensuring that people are informed to set high alarms and loss of connection alarms when using the Medtrum pumps, given the inherent instability in the pump unit itself.

jbr7rr, 2023-10-09:
> Spoke too soon. Overnight last night I suffered the pump stopping working completely, for no apparent reason.
>
> I wonder whether it might be worth ensuring that people are informed to set high alarms [...]

Might be that your pumpbase is defective, I only had this error once and that was with a proper cause; Exceeding the water tightness limitations. Is it ok I send you PB on discord for further information?

tim2000s, 2023-10-09:
Sure.

jbr7rr, 2023-10-09:
> I wonder whether it might be worth ensuring that people are informed to set high alarms and loss of connection alarms when using the Medtrum pumps, given the inherent instability in the pump unit itself.

@MilosKozak Any opinion on forcing the no connection alarm to enabled and 30 minutes for medtrum driver. during init or when it's first activated ?

MilosKozak, 2023-10-09:
if you think it's important do in plugin  `onStart`
```
sp.putInt(app.aaps.core.utils.R.string.key_pump_unreachable_threshold_minutes, 30)
sp.putBoolean(app.aaps.core.utils.R.string.key_enable_pump_unreachable_alert, true)
```

jbr7rr, 2023-10-11:
All things discussed in this ticket are addressed and merged into dev, Imo this can be closed.

tim2000s, 2023-10-16:
> All things discussed in this ticket are addressed and merged into dev, Imo this can be closed.

Agree

closed 2023-10-16 by MilosKozak