nightscout/AndroidAPS #2883: Exponential smoothing back-fill BG gaps (dangerous)

justmara, 2023-10-08:
Today I were changing sensor for my kid and got bad calibration from first attemp, so there was a noticeable gap between last and new BG values. First screenshot shows how (original) Tsunami smoothing that I've shown in that [PR](https://github.com/nightscout/AndroidAPS/pull/2141) handled that on 3.1.0.3 and second one shows how the same gap were handled in 3.2-dev-m with exponential smoothing on my test phone.

As you can see original algorithm simply ignores missing values, while exponential smoothing fills it with non-existing values. That could possibly lead to very dangerous behaviour.

[image]

[image]

ga-zelle, 2023-10-08:
Agreed; glucose data gaps should never be filled be it linearly interpolated or anything else. The fact of the gap is an important bit of information that should be kept in mind. For instance, SMB should be disabled for 10-15 minutes until a new trend is clear.

ga-zelle, 2023-10-08:
That gap filling also happens without using any smoothing option inside AAPS3.2rc1

MilosKozak, 2023-10-09:
filling the gap was always there. in older versions this was not displayed
without interpolation COB calculation would not be working

TranceCake, 2023-10-09:
Isn't it good practise to disable the loop if you don't trust the accuracy of the BG anyway?

justmara, 2023-10-11:
Completed? You've made really scary behaviour and only because noone ever knew It existed before - it is considered _safe_? Wtf?
AAPS must NOT auto-fill gaps with fake BG values. Thats unacceptable.
It can allign existing values to 5-min intervals with interpolation for calculation purpose, but NOT add non-existing values.
Here is an example how BG fall IS interpreted as raise because this fake backfill.

MilosKozak, 2023-10-11:
this behavior is there the since 1st version of AAPS so i'd mark it time proved
and it's there only for calculation purposes, no BG is added

blaqone, 2023-10-27:
> this behavior is there the since 1st version of AAPS so i'd mark it time proved
> and it's there only for calculation purposes, no BG is added

This is not true - it’s there since exp smoothing. And there seem to be a few bad implementations we NEED to talk about!!

sennierer, 2023-12-19:
I realized that in the second graph there seem to be some interpolated data points (around 11) that are real data points in the first graph. Is that intended behavior @MilosKozak ? I realized that in my case (G7, xdrip+ and AAPS 3.2.0.2+autoISF3.0) AAPS seems to replace some of the xdrip values with interpolated ones (at least they appear in the light green color) and its also changing on every loop run which data points are exchanged for interpolated ones.

MilosKozak, 2023-12-19:
@sennierer screenshot?

sennierer, 2023-12-19:
@MilosKozak here are screenshots within the 36h period with the yellow (recalculated data) flag and screenshots after. After the the yellow flag is gone I dont have interpolated data, before I have. xdrip didnt miss any data in both cases.
[image]
[image]
[image]
[image]
[image]

sennierer, 2023-12-19:
and as you can see in the last screenshot delta, delta 15 min and acc are clearly wrong (not sure though if the 0 shown is an actual 0 or rather None or invalid datawise).