nightscout/AndroidAPS #2884: Horrible battery drain with latest dev, stopped using it.

Alamo04, 2023-10-08:
With this dev version my batterie runs out in hours, not days with my old version from 02.09.2023.
Can't go back because its timed out. 😒
[AndroidAPS_LOG_1696772865300.log.zip](https://github.com/nightscout/AndroidAPS/files/12840841/AndroidAPS_LOG_1696772865300.log.zip)

Alamo04, 2023-10-08:
Find the reason,  **it's the DynISF Algorithm**. 

Changed to SMB and battery drain is OK again.

robertrub, 2023-10-10:
I'm not using DynISF and I have a huge battery drain too (list 50% in 6 hours ..).

Cleaned db right now. Will see if it changes anything.

[image]

Alamo04, 2023-10-10:
Now, without DynISF, my batterie is fine.

With DynISF the batterie drains extreme and the phone get very hot. Don't know if that didn't destroy the phone over a longer time.

Will not change back to DynISF until its confirmed fixed.

@robertrub 
For me:
- In Dev, you can clean your db too. Thats enough for me to have no issues with that.
- I deactivated "Automation", in settings. In my case this was also a huge batterie drainer.

MilosKozak, 2023-10-10:
caused by xdrip sync
https://github.com/nightscout/AndroidAPS/commit/0caded7bbb2c09036f4dc20a80b04634eb275441

Alamo04, 2023-10-10:
Thanks @MilosKozak :)

For Info:
Loaded my phone over Night.

After change to SMB yesterday everything was fine, today at 10:00 PM my Phone had 90% batterie left.
Then, in 1,5h it drains to 21%.

Don't understand that.

I have now the new version working, hope the fix works.

Alamo04, 2023-10-14:
@MilosKozak 
After update to latest dev version today, this bug appears again. 3h and batterie empty.
Was using DynISF again.

Going back to the older version (11.10.2023), with your fix. 
That works the last 2 Days, but I was using SMB in that time. Will see what happened with DynISF.

MilosKozak, 2023-10-17:
can you test with latest dev?

robertrub, 2023-10-17:
I just updated. I'm starting testing (using OpanSMB and Automations)

Alamo04, 2023-10-17:
Will do this evening, after work.

Alamo04, 2023-10-17:
Test since 2h, until now ok.

Will report tomorrow evening.

Alamo04, 2023-10-18:
Test after 22 hour's = ok.
If the bug didn't appear in the next 24h, this issue is gone.

Next report tomorrow evening, meantime I will change a little between SMB and DynISF, only to be sure.

Alamo04, 2023-10-19:
Changed this morning to SMB and at noon back to DynISF. Still no batterie issues.

So this issue is fixed and I close it.

closed 2023-10-19 by Alamo04