nightscout/AndroidAPS #2890: Dev M and RC2 AAPS Client missing BG's and regular updates

ohjay73, 2023-10-09:
Tested with Dev M and RC2 and both aaps clients will miss BG updates. All power savings are turned off.. nothing is killing the app. 
Can get updates if i pause websockets for a moment then restart and the data will refresh. 
Regular updates are occurring in NSClient though

[AndroidAPS_LOG_1696874984171.log.zip](https://github.com/nightscout/AndroidAPS/files/12849197/AndroidAPS_LOG_1696874984171.log.zip)
[AndroidAPS_LOG_1696874984171.log.zip](https://github.com/nightscout/AndroidAPS/files/12849201/AndroidAPS_LOG_1696874984171.log.zip)

ohjay73, 2023-10-09:
[image]
[image]
[image]
[image]

szantos, 2023-10-10:
Similar issue here, but I have a slightly older build:
SGV are present in NS, AAPSClient misses to download and/or backfill some of them

AAPSClient 3.2.0-dev-m 81548b7d65-2023.09.29 with NSClientV3 + websockets
NS dev v15: 5f8c734
[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/12853392/Log.Export.txt)

[image]

[image]

MilosKozak, 2023-10-11:
improved in https://github.com/nightscout/AndroidAPS/commit/b870de1e078696a842b934287edf09c7754b313d

closed 2023-10-11 by MilosKozak

ohjay73, 2023-10-12:
@MilosKozak improvements yes.. I can see everything except for BG update now

ohjay73, 2023-10-12:
[image]

olorinmaia, 2023-10-12:
Is NS updated to latest dev? AAPSClient and AAPS? Im feeding AAPS with bg from xdrip and i dont have issues after update yet in AAPS, AAPSClient or NS with sync.

ohjay73, 2023-10-12:
Yes both with latest dev.. I'm using BYOD for BG.. aaps syncs perfectly.. aapsclient will sync then lose bg but gets basal and carbs and smb etc. It'll also send from aapsclient for carbs temp targets etc. Just the BG won't show up unless you pause and restart the connection

ohjay73, 2023-10-12:
[image]

[image]

olorinmaia, 2023-10-12:
Strange. I know what Milos would say;  Create new issue and attach logs :) Just make sure that Nightscout is latest dev also and make sure AAPSclient and AAPS use same API version (stating the obvious) ;)

ohjay73, 2023-10-12:
yup.. added https://github.com/nightscout/AndroidAPS/issues/2905

ohjay73, 2023-10-12:
@olorinmaia how do you send bg to ns? Is it direct from aaps or do you use bridge or the new connect option in NS? I was using bridge during the issue above but am seeing improvements when I switch to aaps direct to ns. Wondering if it's placebo

olorinmaia, 2023-10-12:
I'm using a testrig so setup is like this:
xdrip receive BG from dexcom share.
xdrip local broadcast BG to AAPS.
AAPS upload to NS.

In my live setup with my kid it's like this.
BYODA recieve BG from G6
BYODA local broadcast to AAPS and xdrip
AAPS upload to NS.

I would not recommend using bridge in NS, as it not recommended in live setup either. The way I setup my testrig is as close I can get it to a live setup.

ohjay73, 2023-10-12:
It's looking more and more that bridge and connect are not reliable based on what your setup and what I'm observing with this latest change.

olorinmaia, 2023-10-12:
I would mention your BG setup in the issue. As it probably matters. However, it's a question for @MilosKozak to answer if this is an AAPS bug or NS bug. I don't understand why anyone would want to feed AAPS with BG from NS in a live setup.