nightscout/AndroidAPS #2892: [DynISF] Profile % not affecting ISF

szantos, 2023-10-10:
This is well known for longer term DynISF users. I'd like to post it as an issue before DynISF reaches stable.

Without DynISF a % profile change affects basal, ISF and IC. With DynISF you (only) have the DynISF factor to affect ISF - mainly. TDD also has an effect on ISF, but this is a slow process (due to the TDD weighting of DynISF).

The main threat is confusion: users got used to profile % affecting ISF, too. An example would be exercise preparation with a slightly higher BG, where user sets a lower % profile. Basal gets less, but ISF remains the same, which may be too aggressive.

----

Possible solutions:

1) Leave the behaviour as is, but update documentation (and maybe AAPS app, too) with big red exclamation marks explaining how it works, and the need of adjusting DynISF factor, if doing a % profile change.

2) Let DynISF factor obey profile, as proposed in #2767 (search for obey profile).

3) Put the adjustment factor into the profile, as proposed in #2861

----

Sidenote: Even if 2) or 3) will be implemented, some issues will still remain, as TDD will catch up in 1-7 days, which may cause an unwanted effect on ISF again in the same direction like % profile change. As the TDD mix reacts slowly, this would take days to develop - IMHO this poses a minor issue compared to the current.

swissalpine, 2023-10-10:
The same is true for the other direction: In case of a sudden resistance (disease, cortisone, etc.) the ISF is increased much too slowly.

According to my experiences and my estimation the whole concept is not very convincing, probably works for individuals on "normal" days, but for many not, because numerous cases, which lead for a weaker or stronger insulin supply are not caught and lead to wrong ISF values (especially catheter occlusion, but also bad catheter sites etc.), which are still "wrong" for days.
From this point of view, I fully agree with the above objections ... and would question it as a fundamental extension in the Master at least in the current state of development.

szantos, 2023-10-10:
> numerous cases, which lead for a weaker or stronger insulin supply are not caught and lead to wrong ISF values

Yeah, occlusion or any other event, which can "fake" the TDD poses a risk. At least the user must be aware of this and should be able to take countermeasures (for example deleting bolus from treatments to correct the TDD or switch off DynISF temporary, etc.)

> at least in the current state of development

See also #2893

MilosKozak, 2026-10-11:
Profile percentage now scales the dynamic ISF too (commit 0e16d1885d). A lower % profile switch gives a weaker ISF with DynISF on, the same as without it.

closed 2026-10-11 by MilosKozak