nightscout/AndroidAPS #2893: [DynISF] Dynamically calculated ISF is not used consistently throughout AAPS

szantos, 2023-10-10:
I'd like to post it as an issue before DynISF reaches stable.

There still are many parts of AAPS, where profile_ISF is used, even if DynISF has been selected - this causes confusion and may pose risks:
1 ) Bolus wizard
2 ) COB decay model
3 ) COB (orange) prediction
4 ) UAM (yellow) prediction

DynISF is used for:
5 ) IOB (dark blue) prediction
6 ) ZT (light blue) prediction

As I have only partial understanding of DynISF, I'd like to kindly ask @tim2000s to take a stand, especially regarding 3)-6) - where I'm unsure.

----

Regarding 1) bolus wizard:
As the bolus wizard only uses a single calculation for the end of the DIA, using profile_ISF may result in significant differences compared to the loop calculation.

Example case:
- profile_ISF is tuned for a normal BG level
- BG is high, DynISF calculates a more aggressive ISF, the loop builds up some IOB
- bolus wizard is launched with profile_ISF and thinks the IOB would be too much, and heavily reduces insulin delivered for carbs
- which results in an even greater hyperglycemia
- to avoid this, user may set a more aggressive profile_ISF
- but then the bolus wizard itself may cause a hypoglycemia

As long as the wizard uses another calculation, than the loop, the difference remains. DynISF just leverages it by using another ISF. IMHO there is no easy solution - see suggestion here for implementing the loop algorithm to the bolus wizard: https://github.com/nightscout/AndroidAPS/issues/2561#issuecomment-1701021539

----

2)-6) affect the loop calculation - which is even more complex - but at least one can get used to it and tune the settings accordingly.

----

_I'm using DynISF for more than 1,5 years with hard coded customizations and I'm basically happy with it, as it performs better than pure SMB did. I hope the above contribution will help to make it ready for master - please don't treat it as criticism. Credits go to Chris, Tim, Milos, and co who have brought it to us._

tim2000s, 2023-10-10:
In response to your points:

1. The bolus calculator uses a very basic model and was deliberately not touched when doing dynamicISF. Changing the ISF in that is a decision for @MilosKozak 

2. The COB decay model uses variable sens in the calculation (sens is set to variable sens earlier in the code). The console log still shows profile sens and should be updated. 
3. COB prediction deliberately uses current sensitivity only as that results in a less high prediction and is therefore overall safer in terms of insulin delivery, especially given variability in carb counting and absorption. 
4. UAM uses dynamicISF - line 615 of determine basal in dynamicISF plugin.

olorinmaia, 2023-10-10:
I think there should be more explanation within dynISF plugin what dynISF actually adjusts and what not. To make sure people are aware that profile_ISF still matters for bolus wizard f. ex. This information could be added to where you select DynISF plugin.

There is also one option called "Adjust sensitivity and BG" that needs explanation, as it's only a toggle now, and it's very unclear for me what it really does. Could a string with explanation be added to that option also @tim2000s & @MilosKozak ?

szantos, 2023-10-10:
@tim2000s thank you for the clarification, I'll update the OP with your infos, I'd just like to make sure, I understand it right.

> 1. The bolus calculator uses a very basic model and was deliberately not touched when doing dynamicISF. Changing the ISF in that is a decision for @MilosKozak

Clear.

> 2. The COB decay model uses variable sens in the calculation (sens is set to variable sens earlier in the code). The console log still shows profile sens and should be updated.

Does it mean, it uses DynISF with the recalculation of the ISF at all prediction points?

> 3. COB prediction deliberately uses current sensitivity only as that results in a less high prediction and is therefore overall safer in terms of insulin delivery, especially given variability in carb counting and absorption.

Does it mean profie_ISF?
Or current DynISF (without the recalculation of ISF at all prediction points)?

> 4. UAM uses dynamicISF - line 615 of determine basal in dynamicISF plugin.

Current DynISF?
Or DynISF with the recalculation of the ISF at all prediction points?

MilosKozak, 2023-10-10:
Using Dynamic sensitivity will need more work in core of aaps. This will not fit upcoming release as it will need lot of testing.
It's planned

DFQX, 2023-12-21:
I am using dynISF and reviewed the AAPS documentation. Is there anywhere I can find Chris Wilson related information or reference? Thank you very much!@MilosKozak

DFQX, 2023-12-21:
I am using the DynISF feature. Where can I find information or reference materials related to the Chris Wilson model? Thank you very much!

DFQX, 2023-12-21:
@MilosKozak

olorinmaia, 2023-12-21:
https://androidaps.readthedocs.io/en/latest/Usage/DynamicISF.html

DFQX, 2023-12-22:
@olorinmaia Thanks!  Do you know anything about Chris Wilson's model?

olorinmaia, 2023-12-22:
Not so much, but i know he and Tim is active here on discord! Feel free to ask 😉

https://discord.com/invite/esf8gpF6

DFQX, 2023-12-26:
Thank you very much!

MilosKozak, 2026-10-11:
With DynISF on, the bolus wizard and the COB calculation now use the average dynamic ISF, and the predictions use dynamic ISF (commit a513b657e2). Please open a new issue if you still see a place that uses the profile ISF.

closed 2026-10-11 by MilosKozak