nightscout/AndroidAPS #2903: [Gui enhancement] Mark modified values in profile edit

yossibivas11, 2023-10-11:
I suggest marking lines that were modified (since latest save) in the profile editor, just like we have * in titles of notepad or perhaps a colored background around the modified lines.

kenzo44, 2023-10-14:
@yossibivas11 can you elaborate or upload a screenshot of what you mean.

yossibivas11, 2023-10-14:
[image]
My suggestion is to highlight are otherwise mark any line that was modified

osodebailar, 2023-10-14:
And any suggestion for the light theme ?

yossibivas11, 2023-10-14:
Another option is to change the font style of the numbers to italic

kenzo44, 2023-10-14:
My worry is users might get confused by the highlighted values and assume something is wrong with their profile.

Do you want the highlights to stay after making the changes and activating a profile or just when you edit the profile and hit save?

yossibivas11, 2023-10-15:
Only until I save

olorinmaia, 2023-10-18:
Is it sufficient if text in the box turn **bold**  or _italic_ and maybe add a slightly different text color, but not something that can be assosiated with error. If this happens straight after leaving the box user will know it's because it has been edited. This should also apply to new entries. When you hit save the text in the boxes turn normal.

yossibivas11, 2023-10-18:
> Is it sufficient if text in the box turn **bold** or _italic_ and maybe add a slightly different text color, but not something that can be assosiated with error. If this happens while typing or straight after leaving the box user will know it's because it has been edited. This could also apply to new entries. When you hit save the text in the boxes turn normal.

Exactly what I meant.