nightscout/AndroidAPS #2905: DEV-M AAPSC Client not receiving BG consistently with BYOD and NS15.

ohjay73, 2023-10-12:
When using AAPS Client and BYOD and NS15 all treatments appear except BG. BG will appear at start and sometimes for a short period but will eventually stop and the only way to restart BG is to pause NS client and restart it. All carb entries, SMB, temp targets will flow in both directions but BG has a problem somewhere. 
I am using latest dev as of 10/11 which includes the named commit to fix or improve this issue see pics and logs below. 

Battery optimizations are confirmed off..  
[aapsclientrelease_LOG_1697120441615.log.zip](https://github.com/nightscout/AndroidAPS/files/12883802/aapsclientrelease_LOG_1697120441615.log.zip)
[AndroidAPS_LOG_1697120525944.log.zip](https://github.com/nightscout/AndroidAPS/files/12883803/AndroidAPS_LOG_1697120525944.log.zip)
[image]
[image]
[image]

ohjay73, 2023-10-12:
I was using bridge in NS.. looks like there is a note as of 3.0.0 to disable bridge but i can't find any reasoning for it and its still an active option in any of the NS builds. 

I also tested the CONNECT_SOURCE feature and it uploads the same way as bridge with the same amount of data.

Moving to AAPS upload there is a significant change in the data structure and amount when checking mongo. But it has also eliminated any delay in AAPS Client. 

So.. @MilosKozak is this a bug? is so is it a AAPS or NS bug.. or does it just need clarification in the docs that the AAPS synchronization upload feature is the only one to be used in AAPS 3+?

There was no impact to data in AAPS when using connect, or bridge and when running any reports in NS I didn't see any issues either.. just seems to impact AAPS Client.. and didnt impact NSClient in the past.

szantos, 2023-10-13:
Just an addition: last months I had similar issues (#2748, #2846) and in my setup mainly AAPS was the uploader of SGV.

Regarding NSClient vs AAPSClient - only the name has changed, with ongoing develpoment of course.

MilosKozak, 2023-10-13:
https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html
* While using websockets in NS v3 plugin treatments entered through NS UI (plus button) and other applications using v1 API are not sent to AAPS. This will be fixed in future release of NS. Always use the same client (v1 or v3) in AAPS and AAPSClient until NS fully switch to v3 internaly. The same is valid for AAPS and AAPSClient itself.

that's the reason

closed 2023-10-13 by MilosKozak

ohjay73, 2023-10-13:
@MilosKozak thats for adding entries isnt it?.. which worked in dev - m .. this was related to BG lag in updating aapsclient which would only work if using syncronization from aaps and not bridge or connect_source which worked in previous nsclient builds.