nightscout/AndroidAPS #2906: [DEV] Sync with AAPSClient not working while AAPS screen is off when not charging (V3 API + WS)

olorinmaia, 2023-10-12:
AAPS / AAPSClient 3.2.0-dev-m
Build: 44b8b30637-2023.10.12
Nightscout versjon: DevNS15: b1eb27e88bf01d41c54fe378506dce96876789a9

Strange case so I provide steps to reproduce. AAPS is usually carried with screen off in patients pocket while parent is doing changes from AAPSClient (Temp targets, profile changes, carbs etc)

I noticed today with latest dev, that if AAPS phone screen is turned off and not charging, sync isn't working when enacting e.g. profile switches from AAPSClient

1) turn of screen AAPS phone
2) make sure AAPS don't have charger cable put in
3) enact profile switch from AAPSClient

Result: 
AAPS don't receive profile switch from NS in 7 minutes or so. Profile switch isn't received before I plug in cable for charging or on turn on screen.  

Logs:
[AndroidAPS_LOG_1697132224757.log.zip](https://github.com/nightscout/AndroidAPS/files/12885564/AndroidAPS_LOG_1697132224757.log.zip)

Screenshot:
[image]

MilosKozak, 2023-10-14:
well I thought WS went to sleep by android and rewrote WS to be run in background service (branch `nscv3service`)
but
I realized the main issue was that time on AAPSClient device was ahead of AAPS by seconds and the event was received by AAPS before it's validity. then update was triggered by another event ususally within some minutes
So: please synchronize time on both devices and test again (for me it was working with original dev branch). If you see some issues try the same  with `nscv3service` branch to see if there is an improvement

MilosKozak, 2023-10-14:
it's possible it is working with v1 because it depends on processing on NS side which take some time

olorinmaia, 2023-10-15:
My testrig phone is without simcard (Only Wi-Fi), time settings is android default = same as aapsclient phone with simcard. I can't see any time difference.

It makes more sense to me that v3api+ws should run in background service while ws = off does not to save battery (?).

Fix in branch nscv3service solves the issue in my testrig setup. (Y) :)

olorinmaia, 2023-10-15:
First I double checked that time is same on both devices, minute change in same millisecond and same time settings in android. only difference one of the phones got a simcard. I still experience issue in latest dev.

Seem like nscv3service is an improvement but some time BG in AAPSClient don't update, BG show in NS/NSclient BG but not in overview screen of AAPSClient (need to restart app for it to show.)

AAPSclient screens:
[image]

build: ec849340
AAPSClient logs: 
[AndroidAPS_LOG_1697380195571.log.zip](https://github.com/nightscout/AndroidAPS/files/12909808/AndroidAPS_LOG_1697380195571.log.zip)

AAPS logs:
[AndroidAPS_LOG_1697380229023.log.zip](https://github.com/nightscout/AndroidAPS/files/12909811/AndroidAPS_LOG_1697380229023.log.zip)

MilosKozak, 2023-10-15:
https://github.com/nightscout/AndroidAPS/pull/2925

closed 2023-10-15 by MilosKozak