nightscout/AndroidAPS #2912: Combo (ruffy) error: trying to set TBR=800%

ga-zelle, 2023-10-12:
The pump showed an error "TBR cancelled" while AAPS 3.2rc1 kept interrogating the status and setting TBR=800%. I had to cancel the TBR manuuall a few times until AAPS wanted a TBR below 100%. Thsi never happened before. Using 32.rc1 for about 3 weeks.

In the relevant logfile there are messages that for safety reasons the TBR is reduced 800->500 but sometimes the same safety reasons reduce it from 800->800.
```

```
[AndroidAPS._2023-10-12_02-00-22_.15.zip](https://github.com/nightscout/AndroidAPS/files/12887282/AndroidAPS._2023-10-12_02-00-22_.15.zip)

jotomo, 2023-10-14:
@ga-zelle There are multiple limits applied: the Safety module sanitizes requests and the Combo plugin applies technical limits (max TBR of 500%). Ruffy is never asked to apply a TBR above that limit (grepping the log for 'Attempting to run cmd: SetTbrCommand').
Could you observe this further, I wouldn't know why the code (RuffyScripter hasn't seen any relevant changes in a while) would start to fail. Best guess would be a battery nearing its end and voltage fluctuations causing connnection issues.

ga-zelle, 2023-10-14:
Good point, the battery is nearing its end. Still, it never happened in the last 4 years although I always run the battery until it reports empty.
I intend to move to the new driver anyway before the clocks are set back in 2 weeks time.

closed 2026-10-11 by MilosKozak