nightscout/AndroidAPS #2913: [DEV] AAPSClient misses to download insulin cartridge change

szantos, 2023-10-13:
AAPS 3.2.0-dev-m 44b8b30637-2023.10.13 NSClientV3 + websockets
NS dev v15 [2757fe5](https://github.com/nightscout/cgm-remote-monitor/commit/2757fe585d51dbf6972f26a89291593bee476df3)
[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/12891599/Log.Export.txt)
Time 2023-10-11 23:19

There was an insulin cartridge change, which got uploaded to NS. Probably AAPSClient device was not online that time.
In AAPSClient insulin amount is in sync, but the insulin cartridge change won't update.

[image]

MilosKozak, 2023-10-14:
it's not possible to find the reason from this info. you may try full sync if the event get received

szantos, 2023-10-17:
@MilosKozak
What does full sync do?
Can I mess up something with it in NS?

MilosKozak, 2023-10-30:
https://github.com/nightscout/AndroidAPS/commit/9b4d9b788b1715c7fc662596ceade69e2862ef8a