nightscout/AndroidAPS #2923: Slow database performance in AndroidAPS, especially on history screen

Koeppchen, 2023-10-15:
Using the history screen in the  current dev Version of AAPS is more or less impossible, due to the large number of database statements executed and the database statements being slow. Also, the tdd statistics calculation is slow.
This happens in the dev at least for some month, if not longer (For example dev-m)
I have traced the database statements with  adb shell setprop log.tag.SQLiteTime VERBOSE and then looked at all the statements taking longer then a milliseond with  adb shell logcat | grep "SQLiteTime" | grep "androidaps" | egrep "took [1-9]"
Looking at the database classes i found many indices on columns that have a low number of different values (for tips see https://www.sqlite.org/queryplanner-ng.html#howtofix), for example there are indices on the Table temporary_basals on the fields isValid, pumpType, pumpSerial where in most cases it will be much faster for the database to scan the whole database than using the index.
To check if this is really true i have changed some of the Select statements with hints for the query optimizer. And the sql statements got much faster, the statistices screen with those hints query hints is still slow, but it is usable.
I have tried to change the database itself, but at this point of time i am missing the knowledge how to apply the changes and also for many fields in the database i am not 100% sure about their content.
A will append the diff of my changes to this Issue.

MilosKozak, 2023-10-15:
great. 
history and statistics is querying database but still doing lot of calculations. thus result will never be fast.
I'll try to implement sugestested changes

MilosKozak, 2023-10-16:
https://github.com/nightscout/AndroidAPS/commit/53fb825ae35d46f5ddd34c37c1ca115bf3251902

MilosKozak, 2023-10-16:
it's really significant improvement

Koeppchen, 2023-10-17:
I am sorry, this was not my intention. I could have created the merge request, the same code is more or less in my own branch.
The thing is. there are indices in the Database, for example isValid, that have a very low number of different values. The database uses them to perform the query. Iit would be better to remove those indices instead of using likely/unlikely because without the indices the query is faster in most cases. Examples of "bad" indices are isValid, pumpType, pumpId, pumpSerial, maybe also others.

Koeppchen, 2023-10-17:
And thank you for making it consistent. Better than my diff!

MilosKozak, 2023-10-17:
yes indexes need cleanup too but it will mean increase version number and all consequences. i'm keeping this open to remember it

MilosKozak, 2026-10-11:
The query hints and the cleanup of the low-value indexes (isValid, pumpType, pumpSerial) are both in dev now (database version 34). Thank you for the analysis.

closed 2026-10-11 by MilosKozak