nightscout/AndroidAPS #2936: Minor : time setting window / edited box not the correct color

robertrub, 2023-10-17:
The time setting's window's transparency with dark theme bug have been solved but there is a minor colour "error".

The box I'm modifying (hours below) is the same colour as the background when the box I'm NOT modifying is more "visible".

[image]

It would be better if the box that is being edited has a different colour than the background (black when the background and the other box are dark gray)

osodebailar, 2023-10-17:
The color here is the correct material theme background color.
If you see the light theme, and here there are no special customization done, the background color of the selected edit field is the ticker background. 
So i see here nothing further to do.

robertrub, 2023-10-17:
Ok, if this appearance is the standard, then let's leave it as it is. To me, it seems illogical to _not_ outline where the user is editing but no need to loose sleep over it.

closed 2023-10-18 by MilosKozak