nightscout/AndroidAPS #2938: Nightscout : missing BGs

robertrub, 2023-10-18:
Not sure if it is a NS or AAPS bug... Using last Dev for both.
In AAPS, NS isset to update only when charging. Usually, when charging the last BGs are filled up. They are no more filled. There are holes in the BGs on NS 

[image]

[image]

Trying a full resync of NS in AAPS. 350k entries to update...

robertrub, 2023-10-18:
I see this alert on my NS. I don't know if it's related and why there was this alert.

[image]

Full resync is uploading ok for the moment.

MilosKozak, 2023-10-18:
without logs i cannot resolve anything

robertrub, 2023-10-18:
After a full resync, the holes are there.

Sorry for the logs. Here they are

[image]

You have the approximate times on the graph.

robertrub, 2023-10-18:
It looks like AAPS is uploading only during _in charge_ but doesn't fill up the past BGs.

Setting is G6 to xDrip to AAPS. AAPS does the BG upload to NS.

robertrub, 2023-10-18:
More info : When I set xDrip to upload to NS and to fill in _the past 48h_, all the holes are filled up.

MilosKozak, 2023-10-18:
I need logs 8-10AM

robertrub, 2023-10-18:
Sorry, bizarre but I have a huge hole in the logs too. I'll recreate the situation and will come with newer logs.

MilosKozak, 2023-10-19:
if you are using v3 nothing else using v1( xdrip) must upload to NS

robertrub, 2023-10-19:
Till the "bug" only AAPS was uploading to NS and via API V3.
I used xDrip and v1, just to fill in the holes

MilosKozak, 2023-10-20:
Around 17.20 there are upload errors caused by using xdrip uploads

robertrub, 2023-10-20:
Yes possible as I had used xDrip just before testing to fill in the holes.

Just tested now again (no xDrip connection for hours) but still not uploading.

Will build again and test tomorrow with your last commit on NSCV3 start.

robertrub, 2023-10-21:
@MilosKozak Sorry but your last commit did not solve the problem of filling the past BGs. No xDrip/V1 connection for over 18 hours.

[image]

Look around 9:56 when I reconnected the phone to the charger.

[image]

Seems like this time, there are no more the connection errors alerts I had on my NS page. So the commit solved one problem for sure 😃

MilosKozak, 2023-10-21:
i see data properly uploaded to NS
but i see errors like HTTP FAILED: java.io.IOException: unexpected end of stream on ...
which are related to your NS instance

robertrub, 2023-10-21:
I see that NS Master reverted and updated to 15.0.1 Will update my NS and check again.
BBS

robertrub, 2023-10-21:
Ok, I'm back to NS v15.0.2 Dev.

Good news : the missing BG backfill was a NS bug. Now, when on charge, AAPS is backfilling the BGs. There are no more holes.

The less good news : I got the _Failed authentication error_ on my NS page when AAPS connected :( But the data got updated anyway. I don't know if thid is a NS or AAPS bug... 

Maybe we can close this issue and keep an eye on the _Failed auth error_ problem later in another issue.

closed 2023-10-22 by MilosKozak