nightscout/AndroidAPS #2947: AndroidAPS 3.2 dev hangs after pressing NSClientV3 "switch on" button

basgeurds, 2023-10-19:
Build version 3.2.0-dev-m
Wanted to activate sending data to Nightscout. The option in the Configurator for NSClient can be switched on like normal, but the new NSClientV3 option makes AndroidAPS hang. 
Button turns gray, showing it is selected, but after that nothing happens anymore and the app hangs.
See screenshot.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13044045/AndroidAPS.log)
[image]

olorinmaia, 2023-10-19:
I experienced the same some other day. Not sure how exactly but i made it work again. I think installing rc2 and update to dev again may have helped. Not that it makes any sense 😅 

https://github.com/nightscout/AndroidAPS/issues/2929

MilosKozak, 2023-10-19:
just restart the app helps
it's android issue

basgeurds, 2023-10-20:
Tried restarting the app, but didn't help.
Tried stopping the app and deleting cache data did not help.
Restarting phone didn't help either.

osodebailar, 2023-10-20:
Same problem here on a google pixel 5 with Android 14.

MilosKozak, 2023-10-20:
is after restart NSCv3 activated?

vanelsberg, 2023-10-20:
FYI: Running current dev-m, commit b18f3c9e
Can not repro this problem.

MilosKozak, 2023-10-20:
https://github.com/nightscout/AndroidAPS/commit/2b89a78102afa47cf7d4a3b0a771facd74293ec1

closed 2023-10-20 by MilosKozak

basgeurds, 2023-10-21:
Thank you Milos,
With this update the problem is solved.
Thank you very much for all your work on AndroidAPS!
You help is all a lot, to have a beter life, with better control and less risk of diabetic complications.

Best regards, 
Bas Geurds