nightscout/AndroidAPS #2948: MINOR: New UI buttons change size

robertrub, 2023-10-19:
Not sure if this is a bug or a feature. Since the new rounded UI buttons, when the profile name + percentage is long, the profile button gets bigger, the target button gets smaller.

[image]

I don't think that the buttons changed sizes before. Unless a very long profile name is used, the new presentation is just a bit strange to look at but nothing more than that.

robertrub, 2023-10-19:
In some cases, the target button gets "thicker" which really looks weird

[image]

kenzo44, 2023-10-19:
Will fix this shortly. I did not test with a timed profile change.

closed 2023-10-20 by robertrub