nightscout/AndroidAPS #2950: [3.0.2-rc2] Extended boluses mising on AAPSClient side

vmichalec, 2023-10-19:
[3.0.2-rc2] AAPS, AAPSClient, on both sides NSClientV3 on websockets
[dev, 15.0.0] Nightscout

- I'm using dana-r pump with extended boluses enabled
- In main graph there is a visual difference in rendered basal between AASP and AAPSClient
- On NSClient side in treatments - there is missing extended boluses icon / tab

[image]
[image]

[image]
[image]

And screenshot from nightscout - are they also missing there?
[image]

MilosKozak, 2023-10-19:
this is already fixed in dev
test it and if not mention me here

vmichalec, 2023-10-20:
@MilosKozak
I took 3ffc75bd from this morning, updated Nightscout do 15.0.1 but looks like same. No `NS` sign on such entries  at all. If you need logs, where can I send them?

[image]

[image] [image]

MilosKozak, 2023-10-20:
yes

vmichalec, 2023-10-21:
I have send logs to logs @ aaps.app

MilosKozak, 2023-10-21:
https://github.com/nightscout/AndroidAPS/commit/17c3ef6274f77dac36002ec335a3c8bd201911fc

closed 2023-10-21 by MilosKozak