nightscout/AndroidAPS #2966: [RC2] NSClientV3 FullSync only gets 2 days of TDD.

justmara, 2023-10-22:
Fresh install of AAPS 3.2, NSClientV3, all checkboxes to 'download data from NS', made Full sync. Loaded BG data, treatments, etc, but statistics shows only 2 days of TDD. So DynISF does not work correctly right from start and needs another 5 days to fill the required tdd7 data. While it exists in NS and simply must be downloaded on full sync.
History browser shows BG values, TT changes, notes and sensor change/cannula replacement events for these days, but not bolus/carbs treatments.

[image]

MilosKozak, 2023-10-23:
I cannot replicate this issue. I did v3 setup - clear database - restart app and after v3 finished sync, TDD data was available

justmara, 2023-10-23:
hmm, where did all the screenshots go?
[image]
[image]
[image]
[image]
[image]

justmara, 2023-10-23:
But this time it has loaded 3 days of boluses. And only last portion of boluses from 19.10
[image]

justmara, 2023-10-23:
> what happen when you reset statistics?

exactly same values recalculated

justmara, 2023-10-23:
I see only this fail in log, dunno if it is related
```
16:02:49.404 [RxCachedThreadScheduler-245] D/NSCLIENT: [NSClientV3Plugin$onStart$7.accept():229]: ◄ INSERT TherapyEvent 145
16:02:49.559 [pool-12-thread-1] D/WORKER: [CalculationWorkflowImpl.stopCalculation():111]: Stopping calculation thread: onEventNewHistoryData
16:02:49.576 [DefaultDispatcher-worker-3] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result FAILURE for class app.aaps.workflow.iob.IobCobOref1Worker
16:02:49.580 [pool-12-thread-1] D/WORKER: [CalculationWorkflowImpl.stopCalculation():116]: Calculation thread stopped: onEventNewHistoryData
```

justmara, 2023-10-23:
these are last events from that evening. only last two `0.5` boluses were synced.
[image]

justmara, 2023-10-23:
just made absolutely clean build from RC2 branch in emulator, ran full sync. two days of statistics again.
subsequent syncs does not add anything to this.
still dont understand how did it load 3 days on my phone, but refuses to do the same in emulator (from same NS).
[image]

hartlijn, 2023-10-26:
I have the same. Insulin data seems to get stuck during sync. sometimes (after reset database) it gets a little more data than the other time, but more than 2 days completely doesnt work.

justmara, 2023-10-28:
@MilosKozak this issue can be related https://github.com/nightscout/cgm-remote-monitor/issues/8149
I see many HTTP500 errors during Full sync because NSClientv3 calls treatments/history without `limit` argument

justmara, 2023-10-28:
Seems that I've found the reason.
This request returns ONLY TempBasals and TempTargets
api/v3/treatments?sort=date&**created_at**$gt=2023-10-21T04:28:26.411Z&limit=500

This requests returns Boluses
api/v3/treatments?sort=date&**date**$gt=2023-10-21T04:28:26.411Z&limit=500

Seems to be related to how treatments are sent to NS.

closed 2023-10-30 by MilosKozak