nightscout/AndroidAPS #2968: Bug: Automation list broken after sorting

kenzo44, 2023-10-22:
This bug is present on the master release and the latest dev

Steps to recreate:

- Sort your automations
- When I choose and delete an automation, it deletes the one above or below and not the one I selected
- I cannot delete all automations until I exit aaps and restart it

https://github.com/nightscout/AndroidAPS/assets/4112129/60e37085-bac9-4f46-9f4f-e55a8d6ed946

med06053, 2023-10-22:
I can confirm the issue. It deletes the automation that was previously in that position.
Using dev M.

vanelsberg, 2023-10-22:
> I cannot delete all automations until I exit aaps and restart it
Confirming..

MilosKozak, 2023-10-23:
resolved by https://github.com/nightscout/AndroidAPS/commit/d7e5eea4a3519c93a7d7c80e6efe6f58ab42dc62

closed 2023-10-23 by MilosKozak