nightscout/AndroidAPS #2969: Disabling basal on no BG readings

zibra, 2023-10-23:
Is there any setting to tell loop to disable basal after X minutes of missing BG readings? 

I had recently such case where BG was high because of night pressure, then sensor died for few minutes but basal was still kept high. This led to hypo later.

[image]

MilosKozak, 2023-10-23:
expecting you are not using SMB?

zibra, 2023-10-23:
Yes, I am using SMB

zibra, 2023-10-23:
I know what was wrong. This failure was because of connection issue (device out of range), so AAPS was not able to disable basal because Omnipod was also not in range. I see no issue here. Probably it will be disabled otherwise if it will be just sensor error, right?

MilosKozak, 2023-10-23:
SMB create TBR as long as it expects to be correct. so even when there are not new BGs after issuing TBR, it should result in "good" state

closed 2023-10-23 by zibra