nightscout/AndroidAPS #2974: [Master] Bolus calc doesnt sync through and receive error (v3+WS)

olorinmaia, 2023-10-23:
AAPS / AAPSClient 3.2.0.1
Build: aeed6e2083-2023.10.24
Nightscout version: Version: 15.0.1
GIT-Commit: f46f869c63b3e09f9895c4b99bf596942f5a013c

Bolus from AAPS calc / QW or AAPS wear calc / QW. This instance at 00:47 from wear bolus calc. Sync to NS OK. but not to AAPSClient. Error message in NSCV3-tab.

Happend a few times, attaching logs and screenshots from both AAPS and AAPSClient.

Logs:
AAPSClient: 
[AndroidAPS_LOG_1698101580952.log.zip](https://github.com/nightscout/AndroidAPS/files/13079793/AndroidAPS_LOG_1698101580952.log.zip)
AAPS: 
[AndroidAPS_LOG_1698101501125.log.zip](https://github.com/nightscout/AndroidAPS/files/13079798/AndroidAPS_LOG_1698101501125.log.zip)

Screenshots:
AAPS:
[image]
[image]

AAPSClient:
[image]
[image]

olorinmaia, 2023-10-24:
Tried updating to RC2 or a later dev version I have tested alot with with no errors earlier. And same ERROR (status 400 message parameter limit out of tolerance) is appearing there also.

olorinmaia, 2023-10-24:
I'm using ns.10be.de, and when selecting NS Dev (15.0.2) im not able to reproduce it. But if I redeploy Master (15.0.1) error reappear pretty fast.

MilosKozak, 2023-10-24:
this was a bug in setting in 10be.de
some parameter must adjust

zehnBE, 2023-10-24:
@olorinmaia please try it again with master.
I have patched the lib/api3/generic/collection.js and lib/api3/index.js because it doesnt use the API3_MAX_LIMIT from the my.env on our side (it is in), so i rewrite it to env.API3_MAX_LIMIT.

olorinmaia, 2023-10-24:
@MilosKozak & @zehnBE . Thanks for looking into this. I redployed NS15-master on 10be.de and no errors in nscv3 log. Closing issue.

closed 2023-10-24 by olorinmaia

nl-ruud, 2026-08-15:
Had a similar issue - in my case it was fixed by the provider by updating to Nightscout >= 15.0.7