nightscout/AndroidAPS #2975: Please re-enable insulin button for pen use or during sports if pump is de-activated/disabled

systemcure, 2023-10-24:
I am using AAPS 3.1.0.3 and suffer from missing of insulin button in Home Screen if pump is disabled. Earlier in 2.8. it was always possible to have the insulin button available in Home Screen. Now once the pump is de-activated it is no longer accesible, i.e. giving insulin via Pen is no longer possible while pump is offline. Also during sports I prefer to disable my pump in order to avoid interference from AAPS. But I used the insulin button to force the pump to give me insulin if BG was rising during exercise too much. This lack of insulin button is needlessly restricting the use of pump and pen in these special circumstances and it is causing data loss because I cannot enter my insulin treatments via pen once pump is offline. Since my TDD is very high around >80 IU/day it is necessary that I use Pen for bolusing and just use AAPS for maintaining my Basalrate otherwise injection site will become bad due to too much insulin in the tissue.

MilosKozak, 2023-10-25:
it's disabled because it can be confusing for people trying to send bolus with disconnected pump

MilosKozak, 2023-10-25:
you still can add insuling afterwards