nightscout/AndroidAPS #2977: [Safety Question/Discussion] Change in bolus wizard - 100% Bolus

mSternEU, 2023-10-24:
Not an issue but rather a question that concerns me when I was reading the specific changelog line:
`Change in BolusWizard. If CGM is not available percentage is ignored (ie 100% is used)`

I always get alarmed when I see functionality changed without internal warnings, that comes with the potential of people overdosing when perfoming actions in the same way they are used to before an update.
In this case: Somebody is used to ALWAYS perform 60% bolus via calculator (even without CGM data), and now does perform the same clicks without thinking/noticing, that person will perform an 100% bolus, correct? - Will he/she notice?

I don't doubt there is a good reason for this implementation (I'd like to hear about it), but I'd like to see changes towards "situation with potentially more insulin than expected" being addressed with an in-app warning/confirmation-check

Epic update nonetheless :+1:

jbowler, 2023-10-25:
Please post the URI of the commit, or the commit id, or the actual text of the subject line of the commit.

olorinmaia, 2023-10-25:
Tip: If you enable Engineering mode you can adjust the time needed for this change to occur. 

But I agree that when CGM readings are missing, best practice is to bolus 100%.

mSternEU, 2023-10-25:
> how these users give the missing 40%?

I do understand, that 'no CGM data' for itself is a special occasion. I am just trying to gage whether unintended 'underdosing' or 'overdosing' is the more critical situation to prevent. (Most 'no/wonky CGM data'-situations I witness resolve themselves after 2-3 ticks)

> Tip: If you enable Engineering mode you can adjust the time needed for this change to occur. Commit: [6cd32e8](https://github.com/nightscout/AndroidAPS/commit/6cd32e84fe006bcdfe89a9d1c93aa6e3472f5033)
That indeed is a nice workaround, thanks for that option :+1: 

> But I agree that when CGM readings are missing, best practice is to bolus 100%.
I think I agree with you in general, 100% is fine. I am just worried of situations when default-steps the user is used to suddenly result in higher dosing than expected - without giving him/her a one-time warning during the process.
I'd love to see that kind of precaution / info dialogue whenever routine changes may result in overdosing.

I know I might be biased with overdosing topics, that's probably a side-effect of the amount of kids/parents I supervise and coach. (That's also why I don't like the calculator ignoring TTs per default but that's another issue)
Maybe a 'let users set own default behaviour' option could work.

szantos, 2023-10-30:
Bolusing 100% in this case can cause serious hypos. And if CGM is not coming back in time, the user won't see any declining SGVs or be warned in any way - IMHO a very dangerous situation without any protection.

I hear the argument of dosing the rest. I'd suggest not to touch the bolus delivery %. Instead if there was a % at bolusing and the CGM remains absent after bolusing, there could be delayed notifications (for example after 1-2-3 hours) that the user has to review and dose manually the missing part of the bolus. Or any other solution.

People, who use a % (<100) do it, because they think, the full amount of bolus upfront would be too much and cause early pp hypos (and would like to preserve more agressivity with COB). Overriding this with 100% bolus sets the user towards hypo, and there is no CGM to give any warning.

After a bolus (with CGM data) often TBR=0 is set for longer periods. If CGM is missing AAPS will start to give base basal, which means even more insulin.

@mSternEU , @MilosKozak may it be worth to reopen and reconsider?

MilosKozak, 2023-10-30:
most people uses  <100% because it's recommended and it's how SMB alg was designed. 
not because of possible hypo after bolus but to give to alg working space to adjust dosing based on real cabs income

MilosKozak, 2023-10-30:
and .... without CGM .... it will need user interaction from everyone anyway. maybe in different form but still

MilosKozak, 2023-10-30:
well default value of missing readings is 16 min (and it means in case of dexcom even no BGs for at least next 19 min) and you can change this threshold in engineering mode.
I'm not getting what more do you want

MilosKozak, 2023-10-30:
it you want to update UI just offer something

jbowler, 2023-10-30:
> The setting you are looking for is is: Three-Dot-Menu Preferences Section "Overview" Advanced settings

Ah there is is!  Engineering mode adds another entry below that, "Old glycemia time threshold [min]", with a default of 16 minutes and a minimum of 6 minutes.

closed 2023-11-02 by MilosKozak