nightscout/AndroidAPS #2979: [3.2.0.1] AAPSClient crash because of Pump

justmara, 2023-10-25:
AAPSclient were working fine for about a day after built, but then became crashing on both my and my wife's phones.
Logs shows that Pump driver error is forcing app to exit. How's that possible for aapsclient? 
```
19:46:13.326 [Thread-8] D/PUMPQUEUE: [CommandQueueImplementation.add():184]: Adding: CommandReadStatus - READSTATUS Драйвер помпы изменен.
19:46:13.331 [CommandQueueImplementationHandler] D/PUMPQUEUE: [CommandQueueImplementation.notifyAboutNewCommand$lambda$6():220]: Starting new thread
19:46:13.344 [Thread-28] D/PUMPQUEUE: [QueueThread.run():123]: connection time 0s
19:46:13.346 [Thread-28] D/PUMPQUEUE: [QueueThread.run():129]: performing READSTATUS Драйвер помпы изменен.
19:46:13.350 [Thread-28] D/PUMPQUEUE: [CommandReadStatus.execute():25]: CommandReadStatus executed. Reason: Драйвер помпы изменен.
19:46:13.452 [Thread-28] D/PUMPQUEUE: [QueueThread.run():148]: queue empty. disconnect
19:46:13.455 [Thread-28] D/PUMPQUEUE: [QueueThread.run():152]: disconnected
19:46:13.458 [Thread-28] D/PUMPQUEUE: [QueueThread.run():163]: thread end
19:46:14.969 [main] D/CORE: [MainApp.onCreate():100]: onCreate
19:46:14.996 [DefaultDispatcher-worker-1] D/CORE: [MainApp$onCreate$1.invokeSuspend():116]: Version: 3.2.0.1-aapsclient
```

justmara, 2023-10-25:
Yes, VirtualPump plugin starts invisibli for aapsclient and then at some point sends 'driver changed' event, that crashes app.
This is from aapsclient startup log
```
19:46:15.106 [DefaultDispatcher-worker-1] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: Виртуальная помпа
19:46:15.108 [DefaultDispatcher-worker-1] D/PUMP: [VirtualPumpPlugin.refreshConfiguration():396]: Pump in configuration: Accu-Chek Combo, PumpType object: ACCU_CHEK_COMBO
19:46:15.109 [DefaultDispatcher-worker-1] D/PUMP: [VirtualPumpPlugin.refreshConfiguration():398]: New pump configuration found (ACCU_CHEK_COMBO), changing from previous (null)
19:46:15.110 [DefaultDispatcher-worker-1] D/CONFIGBUILDER: [ConfigBuilderPlugin.loadPref():127]: Loaded: ConfigBuilder_PUMP_VirtualPumpPlugin_Enabled:true
19:46:15.111 [DefaultDispatcher-worker-1] D/CONFIGBUILDER: [ConfigBuilderPlugin.loadPref():133]: Loaded: ConfigBuilder_PUMP_VirtualPumpPlugin_Visible:false
```

justmara, 2023-10-25:
ok, got it - somehow added broken record to `devicestatus` collection - this forces both aaps/aapsclient to crash on receiving it from NS

olorinmaia, 2023-10-25:
Just curious, are you using NSClient v1 or v3 (ws on or off)?  I'm setting up an extra AAPSclient in my testrig to see if I can reproduce.

justmara, 2023-10-25:
I'm on v1 (because changing and applying profile via NS UI)

justmara, 2023-10-26:
seems to be it:
```
java.lang.IllegalStateException: Expected BEGIN_OBJECT but was STRING at line 1 column 25955 path $[26].openaps

com.google.gson.stream.JsonReader.beginObject JsonReader.java:393
com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read ReflectiveTypeAdapterFactory.java:386
com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.readIntoField ReflectiveTypeAdapterFactory.java:212
com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$FieldReflectionAdapter.readField ReflectiveTypeAdapterFactory.java:433
com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read ReflectiveTypeAdapterFactory.java:393
com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.read TypeAdapterRuntimeTypeWrapper.java:40
com.google.gson.internal.bind.ArrayTypeAdapter.read ArrayTypeAdapter.java:72
com.google.gson.Gson.fromJson Gson.java:1227
com.google.gson.Gson.fromJson Gson.java:1137
com.google.gson.Gson.fromJson Gson.java:1047
com.google.gson.Gson.fromJson Gson.java:982
app.aaps.plugins.sync.nsclient.services.NSClientService.onDataUpdate$lambda$14$lambda$13 NSClientService.kt:527
app.aaps.plugins.sync.nsclient.services.NSClientService.$r8$lambda$T6uXFHeO4_jnAJPvIQh8WuxAJy4
app.aaps.plugins.sync.nsclient.services.NSClientService$$ExternalSyntheticLambda0.run
```

MilosKozak, 2023-10-27:
https://github.com/nightscout/AndroidAPS/commit/4dde47bc44562d5082516ca65296b92d318b7773

closed 2023-10-27 by MilosKozak