nightscout/AndroidAPS #2980: Exponential smoothing shows raw value

med06053, 2023-10-26:
Using master version 3.2.0.1 and using exponential smoothing plugin, the graph is smoothed and dynISF plugin uses smoothed values to make calculations, but the value that AAPS shows is the raw value received either by juggluco or xdrip, using libre 2.
Don't know if it is an issue (because it works as intended) or meant to be like that, but in previous versions using the exponential smoothing, the bg value shown by AAPS was the smoothed one.

swissalpine, 2023-10-26:
Same here using G6 and xDrip: The last BG value shown (overview tab) is different to the used smoothed value (SMB tab).
Bug or feature?

MilosKozak, 2023-10-27:
this is how it's currently designed. because of recalculation of dirty data last recalculated(smoothed) value may shift in time too from last received one. I decided to show real value because it represent better overview of what was last value from sensor
recalculated and smoothed values are only in memory and are passed to the alg only

blaqone, 2023-10-27:
I really don’t understand why you would want to show a value that is not being used by the algorithm. We glance at the number to make insulin decisions so I think it should represent the number that is actually used in the calculation process.

MilosKozak, 2023-10-27:
because it can show value as 3 mins old but can be just fresh

blaqone, 2023-10-27:
Sorry i don’t understand your sentence.

blaqone, 2023-10-27:
Together with this problem: https://github.com/nightscout/AndroidAPS/issues/2883 (which was closed without any change) exponential smoothing really is dangerous and showing non consistency.
Showing raw value and filling with non existing values. 

https://github.com/nightscout/AndroidAPS/assets/37814299/b40baedd-6339-46db-b64c-c1ad358793e1

med06053, 2023-10-27:
Maybe a smaller displayed smoothed value could be added to the display, next to the raw value.
Might be less confusing, but I don't disagree with Milos in this question. The raw value might be helpfull in rapid variations.

blaqone, 2023-10-27:
Two different values ne t to each other?!
This is NOT a good idea - the regular user will be really confused. 
Those values should be changeable via toggle - just as you can with a smoothed and raw graph. 

Mixing both makes no sense.

piecycle, 2023-10-27:
Im the author of the exponential smoothing algorithm ("Tsunami smoothing"). The smoothing algorithm itself does just that: provide a smoothed glucose curve to filter out sensor noise which could lead to dangerous loop responses. I also disagree with the showing the raw sensor value to the user. There's a good reason why the smoothed value and not the raw value is passed on to the loop script, and the same reason applies to the user. If a user opens the app and just sees a very high or very low value caused by a noisy sensor, they might be inclined to react even though it's not necessary. Mistakes happen to the best of us, and I think displaying raw values does more harm than good, especially as switching on the smoothing plugin suggest that the displayed value is also smoothed. I don't think this is how it should be and strongly recommend going back to displaying the smoothed value. So far this has never resulted in any problems among the users of my Tsunami mod.

MilosKozak, 2023-10-27:
Checking the actual 3.2 code and smoothed value is displayed
Raw value is used only when bucketed data is not generated (less than 4 recent values available)

MilosKozak, 2023-10-27:
```
    override fun lastBg(autosensDataStore: AutosensDataStore): InMemoryGlucoseValue? =
        autosensDataStore.bucketedData?.firstOrNull()
            ?: repository.getLastGlucoseValueWrapped().blockingGet().let { gvWrapped ->
                if (gvWrapped is ValueWrapper.Existing) InMemoryGlucoseValue(gvWrapped.value)
                else null
            }
```

MilosKozak, 2023-10-27:
https://github.com/nightscout/AndroidAPS/commit/90251178a9a3aa0650ac0717f9008c6784b0fa26
that was it

closed 2023-10-27 by MilosKozak