nightscout/AndroidAPS #2981: watchface shows a question mark instead of a trend arrow

cr582, 2023-10-26:
since the update to 3.2 (both release candidates and master branch), the AAPS watchface often shows a question mark instead of a trend arrow. I have not yet been able to identify a pattern of when this occurs, but it happens with all possible trend directions. It can be that it works in between times for half an hour or so as predicted but then there is again a question mark. See attached picture.

[image]

Philoul, 2023-10-26:
Can you replace AAPS (Digital) by AAPS (custom) watchface (default one or any other one) and send a screenshot?

cr582, 2023-10-26:
[image]

Philoul, 2023-10-26:
I think your issue comes from phone side (trend Arrow sent to the watch is "invalid"...
You will probably have to provide log file to see if we find something in it...
How do you get BG values ?

cr582, 2023-10-26:
through xDrip (dexcom G7). I sent the log through email. Thanks in advance

Philoul, 2023-10-26:
I don't receive log sent by mail. copy it here if you want me to take a look)
But because it's on phone side, I'm not sure I will be able to find root cause without @MilosKozak

cr582, 2023-10-26:
there you go ....

[AndroidAPS_LOG_1698313000571.log.zip](https://github.com/nightscout/AndroidAPS/files/13176000/AndroidAPS_LOG_1698313000571.log.zip)

Philoul, 2023-10-26:
@MilosKozak 
See below an extract of log file, datareceived include a Flat trendArrow on all lines between 11:18:10.446 and 11:18:10.967 but SingleData sent to watch is unknow at 11:18:39.605... very strange
```
11:18:10.446 [main] D/DATABASE: [DataReceiver.onReceive():35]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => -9.806902097696358E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2123071614 23ccbd1-2023.07.16; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Time => 1698311887230; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 78.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G6 Native; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
11:18:10.623 [DefaultDispatcher-worker-4] D/BGSOURCE: [XdripSourcePlugin$XdripSourceWorker.doWorkAndLog():75]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=-9.806902097696358E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2123071614 23ccbd1-2023.07.16, com.eveningoutpost.dexdrip.Extras.Raw=0.0, com.eveningoutpost.dexdrip.Extras.Time=1698311887230, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=-1, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=-1, com.eveningoutpost.dexdrip.Extras.NoiseWarning=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=78.0, com.eveningoutpost.dexdrip.Extras.SourceDesc=G6 Native / G5 Native, com.eveningoutpost.dexdrip.Extras.SourceInfo=G6 Native, com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus=  }]
11:18:10.685 [RxCachedThreadScheduler-613] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():62]: Firing EventNewBG GlucoseValue(id=165883, version=0, dateCreated=1698311890640, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1698311887230, utcOffset=7200000, raw=0.0, value=78.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_G5_NATIVE_XDRIP)
11:18:10.687 [RxCachedThreadScheduler-613] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():67]: Firing EventNewHistoryData 1698311887230 GlucoseValue(id=165883, version=0, dateCreated=1698311890640, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1698311887230, utcOffset=7200000, raw=0.0, value=78.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_G5_NATIVE_XDRIP)
11:18:10.689 [DefaultDispatcher-worker-4] D/DATABASE: [XdripSourcePlugin$XdripSourceWorker.doWorkAndLog():97]: Inserted bg GlucoseValue(id=165883, version=0, dateCreated=1698311890640, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1698311887230, utcOffset=7200000, raw=0.0, value=78.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_G5_NATIVE_XDRIP)
11:18:10.788 [RxCachedThreadScheduler-36] D/NSCLIENT: [NSClientV3Plugin$onStart$9.accept():240]: ► ADD entries Sent PairGlucoseValue <i>{"date":1698311887230,"device":"G6 Native / G5 Native","direction":"FLAT","filtered":0.0,"isReadOnly":false,"isValid":true,"sgv":78.0,"unfiltered":0.0,"units":"MG_DL","utcOffset":120}</i> 165882/165883
11:18:10.808 [OkHttp https://subcutane.ns.10be.de:22704/...] D/HTTP: [NSClientV3Plugin.setClient$lambda$6():348]: {"app":"AAPS","date":1698311887230,"device":"G6 Native / G5 Native","direction":"Flat","filtered":0.0,"isReadOnly":false,"isValid":true,"sgv":78.0,"type":"sgv","unfiltered":0.0,"units":"mg/dl","utcOffset":120}
11:18:10.960 [RxCachedThreadScheduler-612] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():62]: Firing EventNewBG GlucoseValue(id=165883, version=1, dateCreated=1698311890948, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=80fa5dc1-d705-5b58-99d9-d58ba0aad91a, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1698311887230, utcOffset=7200000, raw=0.0, value=78.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_G5_NATIVE_XDRIP)
11:18:10.961 [RxCachedThreadScheduler-612] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():67]: Firing EventNewHistoryData 1698311887230 GlucoseValue(id=165883, version=1, dateCreated=1698311890948, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=80fa5dc1-d705-5b58-99d9-d58ba0aad91a, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1698311887230, utcOffset=7200000, raw=0.0, value=78. [... cut ...] :78.0,"high":165.0,"low":70.0,"deltaMgdl":-2.0,"avgDeltaMgdl":-1.94}
11:18:42.562 [RxCachedThreadScheduler-40] D/WEAR: [DataLayerListenerServiceMobile.sendMessage():203]: sendMessage: /rx_bridge {"type":"app.aaps.core.interfaces.rx.weardata.EventData.SingleBg","timeStamp":1698311881702,"sgvString":"78","glucoseUnits":"mg/dl","slopeArrow":"??","delta":"-2","deltaDetailed":"-2,0","avgDelta":"-2","avgDeltaDetailed":"-1,9","sgv":78.0,"high":165.0,"low":70.0,"deltaMgdl":-2.0,"avgDeltaMgdl":-1.94}
```

MilosKozak, 2023-10-27:
test with this https://github.com/nightscout/AndroidAPS/commit/ff0d327a157cbbf58328be70f0094829d13050e5

cr582, 2023-10-28:
Hi @MilosKozak, I just installed the dev version and will test it for a few days. I'll report back next week. Thanks and regards

cr582, 2023-10-30:
> test with this [ff0d327](https://github.com/nightscout/AndroidAPS/commit/ff0d327a157cbbf58328be70f0094829d13050e5)

it looks like this code change resolved my problem. The trend arrow shows up as expected. No more question marks.

closed 2023-10-30 by MilosKozak