nightscout/AndroidAPS #2982: Upon confirmation of time zone change, received "Insulin delivery is suspended."

birdfly, 2023-10-26:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- 3.1.0.3
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

The story goes like this, not long ago I was traveling across time zones in Europe, and would receive a time zone change notification from AndroidAPS, as shown in the picture below.

[image]

After I clicked on SetTime, an Alert saying "Insulin delivery is suspended" would appear, as shown in the picture.

[image]

What I want to know is, does this notification actually indicate that my Dash Pod has stopped all insulin deliveries, as I see many contradictory pieces of information on the screen?

jbr7rr, 2023-10-26:
Please add logs,

Also check this with 3.2.0.x Not sure if this is solved, but at least the basal information should be correct there.
If you get this error, assume no insulin is delivered until you silence alerts

RBJervey, 2023-11-08:
I can confirm that this happened in my 3.2.0.1 system as well with daylight savings time. About 10-15 minutes after I updated the time in the pod it started beeping at me that insulin delivery was suspended. I went in and silenced the alert and all seemed fine after that.  I will be traveling time zones later this week and will update then (hopefully with logs!)

RBJervey, 2023-11-09:
Logs sent. Traveled from Philly to Phoenix arrived in Phoenix around 2:10pm, pod was beeping by 2:20pm, suspend info in AAPS a few minutes later. Dash-G7-AAPS 3.2.01 and nightscout 15.2.

MilosKozak, 2026-10-12:
Thank you for the report and the logs. This is fixed in the dev branch in 1595ccee47 (it will be in 4.0).

Cause: "Set time" is sent to the pod like a profile change: first a suspend, then a new basal program. The suspend also turns on the pod's "insulin delivery is suspended" alert, which goes off after 20 minutes. After a profile change AAPS turns this alert off again, but after "Set time" it did not. So the pod beeped about 20 minutes later, even though basal was running. AAPS also kept the "pump suspended" temp basal open, which is why the screen showed mixed information.

Now "Set time" is handled like a profile change: the alert is turned off and the suspend temp basal is ended.

We could test this only with unit tests, not with a real pod. @birdfly @RBJervey, if you use a dev build, could you please test it at your next time zone or DST change:
1. Confirm "Set time" in AAPS.
2. Wait at least 25 minutes. The pod should not beep.
3. The Dash history should show "Set time" followed by "Configure alerts", and AAPS should not show the pump as suspended.

If it still beeps, please comment here with the logs and we will reopen the issue.

closed 2026-10-12 by MilosKozak