nightscout/AndroidAPS #2985: 3.2.0.1 new APK res/q-.apk duplicates many files in res/

jbowler, 2023-10-26:
This is a Hedgehog fullRelease build of aeed6e208393a6dd0c43c58c17a2529f6a646de4

The new APK seems to be a complete version of the res/ directory with some changes.  It more than doubles the disk size of the res directory itself and results in about 10MByte more in the 3.2.0.1 APK compared with 3.2.0-rc2 and previous releases.  From AndroidStudio analysis/guesstimates:

3.2.0-rc2: APK size: 55 MB, Download Size: 20.9 MB
3.2.0.1: APK size: 66.2 MB, Download Size: 30.5 MB

So it may also increase the download size by 50%, though that's not relevant in practice.

MilosKozak, 2023-10-27:
no idea where is this coming from

jbowler, 2023-10-27:
Here (from the StudioProjects directory I built a couple of day ago):

jbowler@hippopopus ~/StudioProjects/AndroidAPS-3.2 $ sh -x ../ttt
+ find . -type f -print0
+ xargs -0 md5sum
+ egrep 7ee9d9e80f9d664ca73bfe72f18ce53f
7ee9d9e80f9d664ca73bfe72f18ce53f  ./app/build/intermediates/packaged_res/fullRelease/raw/android_wear_micro_apk.apk
7ee9d9e80f9d664ca73bfe72f18ce53f  ./app/build/generated/res/microapk/full/release/raw/android_wear_micro_apk.apk

It's not in the git source (I checked in the same way) but somehow the build now puts a, for me, spurious APK into the resources.

MilosKozak, 2023-10-27:
[image]
I don't see  wear apks in final apk

jbowler, 2023-10-27:
That's it.  SteamPunk.zip also appears to be watch related ("CustomWatchface.png" for example).  I don't know how to check the signature on q-.apk but that might give a clue.  Also the build log presumably shows what built it (I don't know enough AndroidStudio to even find the build log.)

MilosKozak, 2023-10-27:
the change between rc2 and 3.2.0.1 is migration to kts builds. maybe there is something different. but i didn't find a option how to affect it

closed 2024-01-18 by jbowler