nightscout/AndroidAPS #2990: 3.1.0.3 + 3.2.0.1 No Pairing possible

sfriedriszik, 2023-10-27:
The problem here is that in the pairing process the pump goes into communication according to the pairing codes,
the arrows appear but then the process stops. 

The whole thing happens reliably 12 times in a row. It is a DANA RS V3 from 03/2023. 
Another older RS in working well.

Log attached.
[AndroidAPS_LOG_1698424097983.log.zip](https://github.com/nightscout/AndroidAPS/files/13191320/AndroidAPS_LOG_1698424097983.log.zip)

MilosKozak, 2023-10-27:
unpair pump in android
reset pairing on Dana tab by holding BT icon
then try again

MilosKozak, 2023-10-27:
or .... try to pair the pump to another phone first and then back

sfriedriszik, 2023-10-28:
@MilosKozak 
tried both approaches. still without success.
Is there another approach?

(log attached)
[AndroidAPS_LOG_1698510482597.log.zip](https://github.com/nightscout/AndroidAPS/files/13196327/AndroidAPS_LOG_1698510482597.log.zip)

MilosKozak, 2023-10-30:
try original anydana app

sfriedriszik, 2023-10-30:
@MilosKozak Anydana App works.

MilosKozak, 2023-10-30:
hmmm sorry,  no more ideas
is any other pump with the same firmware working?

MilosKozak, 2023-10-30:
does the dialog for entering keys appear on screen?

sfriedriszik, 2023-10-30:
@MilosKozak 

Yes, the dialog appears and the values can also be entered and confirmed.
then the arrow graphic appears and then the process stops.

MilosKozak, 2023-10-30:
then  i don't know

sfriedriszik, 2023-10-30:
@MilosKozak 

Anyway, thank you very much for your efforts

mSternEU, 2023-11-02:
Issue detected/solved, it was a display resolution problem.

The pairing dialogue doesn't get scrollable when showing more content than the current device resolution allows, so the user isn't able to reach the "cancel / ok" buttons (or can't even guess they are there).
Implementing scrollable diaglogue, fix-binding exit/confirmation-buttons to the visible bottom part or allocate the Enter-Key / bottom right key of the virtual keyboard to a 'send' command could be helpful enhancements.

This is a "small device/screen"-issue, like Jelly/oldphone users.

Workaround was "connecting pump via big-screen phone and import saved settings to Jelly" 

(Screen was taken from a working phone though, just as a reference)
[image]

MilosKozak, 2023-11-02:
https://github.com/nightscout/AndroidAPS/commit/bfacfd925590d458d08fd29ddbf09dcc3fd91996

closed 2023-11-02 by MilosKozak

mSternEU, 2023-11-02:
Beautiful, much appreciated, thank you :clap: