nightscout/AndroidAPS #2992: bg target not aligned on client

painbrain81, 2023-10-27:
in the 3.2.0.x and 3.1.0.x client app the target always remains that of the active profile. on androidaps master it is varied with autosens but on the client it always remains the same. Is it a mistake or is it intentional?
on client app it changes only if temp-target enable (from master or from client, it's the same)

MilosKozak, 2023-10-27:
can you provide screenshot?

painbrain81, 2023-10-28:
Sure! As soon as there is an opportunity for a target modified by autosens I will immediately take screenshots. Thank you.

justmara, 2023-10-28:
There were options `sensitivity rises target` and `resistance lowers target` - only those could affect target by sensitivity. Do you have those enabled?

painbrain81, 2023-10-28:
Yes. They are enabled and in fact on master it's ok (raise and decrease for autosens). In NSclient i see always only target from current profile.

painbrain81, 2023-10-29:
> can you provide screenshot?

here is 3 screenshots. Master, NSclient and OPAS detail popup in NSclient.
[image]
[image]
[image]

MilosKozak, 2023-10-30:
https://github.com/nightscout/AndroidAPS/commit/aa2ae1b99fe9dc6c02338bb86495ce0cc93093f9

closed 2023-10-30 by MilosKozak