nightscout/AndroidAPS #2993: Connection settings for NS and Tidepool

painbrain81, 2023-10-28:
The current situation is that connection settings are shared between NS and Tidepool. This makes it impossible, for example, to upload data to NS only with mobile data and Tidepool data only with WiFi. Is it possible to separate options for each uploader?

painbrain81, 2023-10-28:
my kid is T1 and i absolutely need Nightscout always on. 
...instead of tidepool which can safely upload data via wifi when charging (less battery in poor words)

MilosKozak, 2026-10-10:
Done in 649e4abbb8, for AAPS 4.0.

Tidepool settings → Connection settings now has **"Use Nightscout connection settings"**. It is on by default, so nothing changes for current users. Turn it off and Tidepool gets its own settings: cellular data, roaming, WiFi, WiFi SSID, sync on battery, sync while charging.

Example: Nightscout always on (cellular too), Tidepool only on WiFi while charging. Closing.

closed 2026-10-10 by MilosKozak