nightscout/AndroidAPS #2994: It is unclear where the entry Carbs came from

KochiaX, 2023-10-28:
Last night at 2.15 it is unclear from where there was a record of 96 grams of carbs. 96 is the max allowed carbs according to the settings. Around this time, I made a bolus for correction of 1.5 units through the button Insulin, but carbs were not introduced. As a result, AAPS started injecting SMB, which led to hypoglycemia.
I use AAPS 3.1.0.3, Nighscout 14.2.5.
[image]
[image]
[image]

MilosKozak, 2023-10-28:
i can tell you if you have logs

KochiaX, 2023-10-28:
> i can tell you if you have logs

Are these logs?

[AndroidAPS._2023-10-27_00-00-01_.0.zip](https://github.com/nightscout/AndroidAPS/files/13195815/AndroidAPS._2023-10-27_00-00-01_.0.zip)
[AndroidAPS._2023-10-27_00-00-01_.1.zip](https://github.com/nightscout/AndroidAPS/files/13195816/AndroidAPS._2023-10-27_00-00-01_.1.zip)

vanelsberg, 2023-10-28:
Maybe wrong button pressed?
Your logs show user pressing button for Carbs entry:

```
02:16:15.959 [main] D/APS: [DialogFragmentWithDate.onViewCreated$lambda-7():130]: Submit pressed for Dialog: info.nightscout.androidaps.dialogs.CarbsDialog
....
02:16:18.377 [RxCachedThreadScheduler-52] D/CORE: [UserEntryLogger$log$2.invoke():45]: USER ENTRY: CARBS CarbDialog  [Gram(value=96)]
```

MilosKozak, 2023-10-30:
exactly, someone entered them through Carbs dialog

closed 2023-10-30 by MilosKozak