nightscout/AndroidAPS #2995: Eopatch/Glucomen Day pump connection problem phone pixel 7a

loopzwenneke, 2023-10-28:
Bluetooth communication problem when trying to connect Glucomen day pump with pixel 7a.
New pod isn't shown when trying to connect. Also working pod on another phone could not be transferred to pixel 7a . Reporting "communication problem". Emptied bluetooth cache , didn't help
AAPS 3.2.0.1

[AndroidAPS_LOG_1698487464169.log.zip](https://github.com/nightscout/AndroidAPS/files/13195483/AndroidAPS_LOG_1698487464169.log.zip)

jbm76, 2023-10-29:
> Bluetooth communication problem when trying to connect Glucomen day pump with pixel 7a. New pod isn't shown when trying to connect. Also working pod on another phone could not be transferred to [...]

The exact same thing just happened to me With the latest version 3.2.0.1, and with two new ones from EoPatch and on two different phones, Pixel 7 and Oneplus Nord 3.

Now reinstalling the dev-m.

MilosKozak, 2023-10-29:
I'm able to pair EOP2 wint S10e and 3.2.0.1

jbm76, 2023-10-30:
> I'm able to pair EOP2 with S10e and 3.2.0.1-dev

Thank you very much for your verification @MilosKozak !!

What I did see, when changing to another phone, is that when importing my previous profile/settings (json), data is brought from the pod that I had when I exported the json file. (example: insulin used, insulin remaining, days consumed from the pod)

Is this normal? Could it be bringing data from the old pod and causing problems when pairing the new one?

Thank you very much again for everything and greetings

zwoes, 2023-10-30:
Thanks for commenting. I'm still able to pair EOPatch/Glucomen day pods with older Sony Xperia , ...as before (dev branches and AAPS 3.2.01), but would like to pair with newer pixel 7a phone...
Couldn't pair before either (dev branches) with pixel 7a and therefore tried with the new AAPS 3.2.0.1 ...without success...Don't know if this is because of specific bluetooth characteristics of pixel7a?

closed 2023-11-15 by zwoes