nightscout/AndroidAPS #2998: Master 3.2.0.1 Winter time open loop

robertrub, 2023-10-29:
I got a message in AAPS telling that the loop was not closed ad the winter time change was in 3 h (or something similar, I was half asleep...)

But, the UI showed the closed loop icon

[image]

And when I looked in the OpenSMB tab, nothing was selected (not even open loop)

[image]

And, strangely, I didn't get the usual "update time in DADH pod" messages....

mtanjagithub, 2023-10-29:
Check the docs.
Other pumps
This feature is available since AAPS version 2.2.

To prevent difficulties the Loop will be deactivated for 3 hours AFTER the
DST switch. This is done for safety reasons (IOB too high due to duplicated
bolus prior to DST change).

You will receive a notification on the main screen prior to DST change that
loop will be disabled temporarily. This message will appear without beep,
vibration or anything.

https://androidaps.readthedocs.io/en/latest/Usage/Timezone-traveling.html#time-adjustment-daylight-savings-time-dst

robertrub, 2023-10-29:
Except the point above, it was the easiest summer/winter time change ever. No need to set pod time, no other messages etc :)

robertrub, 2023-10-29:
@mtanjagithub  Yes, I know the drill and the reason. That is NOT the "bug".

The strange/buggy (?) part is the icon, it says closed loop... And the loop selection just disappears.

Maybe adding a barred TZ or DST on the closed loop icon to represent the situation correctly would be better (for those who wake up in the middle of the might ;) )

olorinmaia, 2023-10-29:
My input:
I couldn't see a reason in SMB / DynISF tab tonight why loop wasnt working, maybe I was to tired to see where it was tho. So at first I thought it was a bug with DynISF that is new for us ;)

I had to see inside "Loop"-tab to see error message, "DST helper: DST in last 3 hours" isn't really self-explantory either. I can't remember getting any warning in AAPS. 

Loop icon should turn red (disabled) and SMB/dynisf-tab should also show error why loop isnt working during these 3 hours. That may be easy to fix.

robertrub, 2023-10-29:
```
 couldn't see a reason in SMB / DynISF tab tonight why loop wasnt working
```

Normally you had a message in AAPS telling that loop can't ve closed due to DST...

But agreed on the modified icon.

closed 2024-12-15 by robertrub